jetperch/pyjoulescope_ui #2: Multimeter displaying NaNs after running overnight

JayKickliter, 2019-06-28:
The screenshot tells it all:
[image]

mliberty1, 2019-06-28:
Thanks for creating an issue, but maybe it doesn't quite tell all ;)  I have a few questions:

1. Are the remaining fields still updating, or are they "frozen" at the values shown?
2. What is your OS and version (looks like Mac OS)?
3. Do you know if your computer went into sleep mode overnight?

Would you mind attaching the Joulescope file log for this run (the filename is UTC time for the application start)?  You can find it under:

    ~/.joulescope/log/

Thanks!

JayKickliter, 2019-06-28:
> Are the remaining fields still updating, or are they "frozen" at the values shown?

They are still updating, including plots.

> What is your OS and version (looks like Mac OS)?

macOS 10.14.5

> Do you know if your computer went into sleep mode overnight?

shouldn't have, but I can't guarantee it.

### Logs

[joulescope_20190626_180736_3378.log](https://github.com/jetperch/pyjoulescope_ui/files/3339912/joulescope_20190626_180736_3378.log)
[joulescope_20190627_150054_1471.log](https://github.com/jetperch/pyjoulescope_ui/files/3339913/joulescope_20190627_150054_1471.log)
[joulescope_20190627_183238_2584.log](https://github.com/jetperch/pyjoulescope_ui/files/3339914/joulescope_20190627_183238_2584.log)

mliberty1, 2019-06-28:
Thanks @JayKickliter!  The last log file joulescope_20190627_183238_2584.log is the correct one for the overnight run.  I am not sure what happened as the log does not show anything abnormal other than a few missing samples at 11:58, 12:25 and 8:16.   I will try to duplicate this.

JayKickliter, 2019-06-28:
Thanks for looking into it. It's barely an inconvenience for me, but other customers might rely on long runs. 

As an aside, everyone in my office is impressed with the Joulescope after just one day, and we'll probably be ordering more.

mliberty1, 2019-06-28:
Great to hear that you are finding Joulescope useful!

The NaNs on the multimeter view should not happen, and I will definitely investigate.

heroichornet, 2019-07-09:
Hi, I can confirm this issue
[image]

All I did was let it run from around 12 hours over night.

mliberty1, 2019-07-09:
Hi @heroichornet and thank you for the posting your confirmation of this issue.  Could you check what version you are running (click Help → About)?  I fixed two related defects ([here](https://github.com/jetperch/pyjoulescope/commit/b15be0358940f9db4fda281ba57856868008e4a1#diff-f646faa0694aed393944d8afae88375a) and [here](https://github.com/jetperch/pyjoulescope/commit/e75a0d00f815529d3c4bc9807f71658812cc538f#diff-f646faa0694aed393944d8afae88375a)) that were the likely contributors to this issue.  Release 0.4.5 contains these fixes, and I ran 0.4.5 for 2.5 days on Win 10, Ubuntu 18.04 and macOS 10.14.5.  I no longer saw this issue, so I think it is fixed.  However, I have not yet created a good regression test, and I did not try Win 7.  If you are not running 0.4.5, could you upgrade to [0.4.5](https://www.joulescope.com/download/) and see if that fixes the issue?

Under Win 7, the Joulescope log files are located in:

    {your_appdata_path}\Local\joulescope\log

For me with username mliberty, the full path is:

    C:\Users\mliberty\AppData\Local\joulescope\log

heroichornet, 2019-07-09:
@mliberty1 Thank you for the quick reply

 I am running 0.4.5
[image]

mliberty1, 2019-07-09:
@heroichornet If you can post the log for the NaN run, I would appreciate it.  The logs are named based upon the UTC timestamp.  I will also try to duplicated using my Win 7 VM.

heroichornet, 2019-07-09:
[joulescope_20190708_081558_8556.log](https://github.com/jetperch/pyjoulescope_ui/files/3373814/joulescope_20190708_081558_8556.log)

heroichornet, 2019-07-11:
I just tried with the small test script in the "Developer" section that calculates the mean of 0.25 seconds of measurements and the output was also similar with "nan" average

$ python test.py
[[0.01208843 3.3185446 ]
 [0.01210787 3.31968   ]
 [0.0121348  3.31968   ]
 ...
 [0.00943616 3.31968   ]
 [0.00942718 3.31968   ]
 [0.00940175 3.3174095 ]]
nan A, nan V
(joulescope)

Maybe that helps

mliberty1, 2019-07-11:
@heroichornet Thanks for taking a closer look.  The example code you found in usb_inrush.py is essentially doing a nanmean by filtering to include only finite (no NaN, +INF, -INF):

    valid = np.isfinite(i_mean)
    voltage = np.mean(v_mean[valid])

I should take another look at that usb_inrush.py code as I doubt the USBET tool handles NaNs nicely.

NaNs are injected into the Joulescope sample data whenever samples are missed.  This happens frequently at the start of data streaming.  Dropped samples should happen very infrequently once streaming is ongoing.  However, they do happen regardless of USB transfer mode.  In USB BULK mode (which the Joulescope software uses today), data integrity is guaranteed, but throughput is not guaranteed.  The host should be able to keep up.  I could switch to using USB ISOCHRONOUS mode, which guarantees throughput but not data integrity, so we could still get some corrupted USB frames = missing samples.  I now have a setup under a Win 7 VirtualBox VM that drops data all the time (the USB VM adapter just doesn't seem to keep up), so I can better observe this case.  I'll be doing more testing today & tomorrow.

The software is designed to handle missing samples as best as possible, but I found & fixed another issue that could be contributing (pyjoulescope [bc50129](https://github.com/jetperch/pyjoulescope/commit/bc50129aa1e84b39f9281c4c011f4f0a6fa1dadf)).  

If you just want a capture containing contiguous samples, the driver has a contiguous_duration option.  Also see the "--contiguous" option for the command-line capture command:

    python -m capture --help

I am adding an indicator to the UI that will more clearly show if your system is dropping samples.  Right now, you have to look at the log or open the developer View.  You can look at continguous_count which is the number of contiguous samples in the current block.  You can also compute the dropped fraction:

    dropped_fraction = sample_missing_count / sample_id

mliberty1, 2019-07-28:
The 0.5.0 release should fix this issue.  Could you try upgrading and let me know if you still observe this issue?  Thank you!

heroichornet, 2019-07-28:
I will upgrade and report back. I actually have not seen the issue since upgrading to 0.4.6

heroichornet, 2019-07-30:
I let 0.5.0 run over night and was unable to reproduce this issue. I think you can close this issue.

mliberty1, 2019-07-30:
I will close the issue since we audited the code and NaN contributors.  Although we think we got them all, it's hard to be 100% confident with NaNs.  If you do happen to see NaNs again, please reopen this issue or start a new issue.  Thank you @heroichornet and @JayKickliter.

mliberty1, 2019-09-24:
Hi @pawelm87 - thank you for reporting this issue.  What likely happens is that your mac goes off and does something that causes Joulescope sample drops.  The UI should handle this better than displaying NaNs.  It looks like you have pressed the "Accumulate" button.  I inspected the code, and I found that this accumulate function does not handle NaNs well.  We will fix this.

If you want to confirm that your Mac dropped Joulescope samples, you can either inspect the log yourself or post it here, and I'll take a look.  On Mac, you should find Joulescope logs under ~/.joulescope/log, and the file is the starting time in UTC.

pawelm87, 2019-09-25:
[joulescope_20190923_113000_94106.log](https://github.com/jetperch/pyjoulescope_ui/files/3651359/joulescope_20190923_113000_94106.log)
[joulescope_20190924_061401_57851.log](https://github.com/jetperch/pyjoulescope_ui/files/3651361/joulescope_20190924_061401_57851.log)
[joulescope_20190923_112946_94099.log](https://github.com/jetperch/pyjoulescope_ui/files/3651362/joulescope_20190923_112946_94099.log)

mliberty1, 2019-09-25:
Hi @pawelm87 and thank you posting the log files.  The overnight log files are:
joulescope_20190923_113000_94106.log
joulescope_20190924_061401_57851.log

Both show that your computer went through a few time periods where it did not keep up with the 
USB data stream.  All of those log entries:
   joulescope.stream_buffer:Fill missing samples
indicate that your mac is no longer keeping up with the Joulescope USB data rates.  These sample drops means that the Joulescope software never receives this data resulting in inaccurate values being displayed.  I suspect that your computer is entering a low power mode, but I'm not exactly sure.  

Here are some things to try, if you haven't already:
1. Ensure that your mac is powered from the wall (not just battery)
2. Select **System Preferences** → **Energy Saver** → **Power Adapter**.  
   a. Check "Prevent computer from sleeping automatically when the display is off"
   b. Uncheck "Enable Power Nap while plugged into power adapter"
   c. Click **Schedule..." and ensure that no schedules are set.

The upcoming 0.6.4 software release will fix the NaN display, but samples will still be dropped on your computer without any changes.

pawelm87, 2019-09-30:
So, if the Joulescope miss samples that means the power consumption which is display is also inaccurate values

I will check setting in Mac

pawelm87, 2019-09-30:
[image]

I uncheck  "Enable Power Nap"
No Schedule is set

When I left for the night, I used "screen lock" on the Mac

pawelm87, 2019-09-30:
I changed all the settings on the Mac you wrote about, and again NaN. Mac wasn't powered from battery.

I need to test device for long time and meassure power consumption

mliberty1, 2019-09-30:
Hi @pawelm87 - How long do you plan on performing testing?  For very long-term testing, I recommend using the [downsample logging](https://github.com/jetperch/pyjoulescope_examples/blob/master/bin/downsample_logging.py) script.  For very long-term testing, you should also disable network and all other software to minimize the chance of things going wrong.

I also thought of something things to check:

1. Please uncheck "Put hard disks to sleep when possible".
2. For highest reliability, please ensure that your Joulescope is plugged directly into your computer's USB port (no hubs or docks).  Note that many USB-C adapters are hubs, so you just want to use a USB C to USB A cable.
3. How much total and free RAM does your system have?  Open finder → **Applications** → **Utilities** → **Activity Monitor**.  Click **Memory**.  You should have at least 1 GB free for the Joulescope UI.  I measure that it uses 786.5 MB on my 2018 13" MacBook Pro with macOS Mojave 10.14.6.  Also check on system CPU utilization.  I noticed that 0.6.3 does use quite a bit of CPU (about 60% from my) on mac, even when just in multimeter view.  I will investigate.

pawelm87, 2019-10-02:
Hi @mliberty1  I tryed run this script but I have got error :

`Installing collected packages: pymonocypher, pywin32, pypiwin32, joulescope, pyparsing, kiwisolver, cycler, matplotlib
  Running setup.py install for pymonocypher ... error
    ERROR: Command errored out with exit status 1:
     command: 'c:\users\p\appdata\local\programs\python\python37-32\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\PAWEL~1.MAR\\AppData\\Local\\Temp\\pip-install-5fuvcsct\\pymonocypher\\setup.py'"'"'; __file__='"'"'C:\\Users\\PAWEL~1.MAR\\AppData\\Local\\Temp\\pip-install-5fuvcsct\\pymonocypher\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\P\AppData\Local\Temp\pip-record-2gkf8n3s\install-record.txt' --single-version-externally-managed --compile
         cwd: C:\Users\P\AppData\Local\Temp\pip-install-5fuvcsct\pymonocypher\
    Complete output (5 lines):
    running install
    running build
    running build_ext
    building 'monocypher' extension
    error: Microsoft Visual C++ 14.0 is required. Get it with "Microsoft Visual C++ Build Tools": https://visualstudio.microsoft.com/downloads/
    ----------------------------------------
ERROR: Command errored out with exit status 1: 'c:\users\p\appdata\local\programs\python\python37-32\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\P\\AppData\\Local\\Temp\\pip-install-5fuvcsct\\pymonocypher\\setup.py'"'"'; __file__='"'"'C:\\Users\\P\\AppData\\Local\\Temp\\pip-install-5fuvcsct\\pymonocypher\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\P\AppData\Local\Temp\pip-record-2gkf8n3s\install-record.txt' --single-version-externally-managed --compile Check the logs for full command output.`

I have installed Microsoft Visual C++ Build Tools

pawelm87, 2019-10-02:
Ok I install all dependencies and run this script:

```
PS C:\Users\p\Desktop\workspace\Joulescope\pyjoulescope_examples\bin> py.exe .\downsample_logging.py
Starting logging - press CTRL-C to stop
Filename: C:\Users\p\Documents\joulescope\jslog_20191002_124932_20452.csv
2019-10-02T14:49:32 WARNING  No Joulescope connected
2019-10-02T14:49:43 WARNING  No Joulescope connected
PS C:\Users\p\Desktop\workspace\Joulescope\pyjoulescope_examples\bin>
```

Joulescope is connect and all app which will be use it is closed

mliberty1, 2019-10-02:
Hi @pawelm87 - Can you use 64-bit Python?  We do not support 32-bit python.  You can follow the [Quick Start Guide](https://github.com/jetperch/pyjoulescope#quick-start).  Once the command

    python3 -m joulescope capture --contiguous 1.0 mycapture.jls

works, then you can try downsample_logging.

pawelm87, 2019-10-02:
ok work now i will test it

mliberty1, 2019-10-03:
Following up on this issue, I ran for nearly Joulescope UI 0.6.3 in multimeter view for 3 days on my 2018 13" MacBook Pro with all power management settings disabled.  No NaNs.  With the upcoming Joulescope UI 0.6.4 (available today or tomorrow), you will no longer see NaNs, but you may see sample drops.  Sample drops are currently only visible in the **Developer** view.

[image]

I am going to close this issue, but feel free to reopen if you still see NaNs in 0.6.4.

closed 2019-10-03 by mliberty1