jetperch/pyjoulescope_ui #3: Ui_DeveloperDockWidget is missing

ferret-guy, 2019-06-30:
When trying to run from source I found that the following widgets are missing:

- Ui_DeveloperDockWidget
- Ui_ErrorWindow
- Ui_mainWindow
- Ui_ControlDockWidget

mliberty1, 2019-06-30:
These missing files are expected since they are generated from the corresponding Qt ".ui".  For now, run:

    python3 setup.py sdist

which generates these files.  I will investigate and update the README with the recommended way of generating these files for development.

ferret-guy, 2019-06-30:
Thanks for the quick response! I am unfamiliar with Qt so I appreciate the help.

mliberty1, 2019-06-30:
No worries!  I looked and I haven't documented this.  "compiling" before running Python code is very unusual, so I need to figure out how to make this easy...

mliberty1, 2019-07-08:
I added a separate "qt" command so that you can just generate (or regenerated) the Qt resource files and the Qt python files from the ".ui" files.  Run:

    python3 setup.py qt

This change is in develop now, and will be merged into master in the next release.

closed 2019-07-08 by mliberty1