jetperch/pyjoulescope_ui #5: Error with pip install

jeremyherbert, 2019-07-08:
I'm trying to install the joulescope-ui on OSX high sierra. I get the following error when using `pip3 install joulescope-ui`

```
Collecting joulescope-ui
  Downloading https://files.pythonhosted.org/packages/c5/cb/09bd4abd9645199378453501f3d18b955329199f9729880c0f18624e219b/joulescope_ui-0.4.5.tar.gz (155kB)
    100% |████████████████████████████████| 163kB 1.6MB/s
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/private/var/folders/d2/zsp27n4j6sg3mpr9zztg2w0m0000gn/T/pip-install-ak5j6mhr/joulescope-ui/setup.py", line 33, in <module>
        from pyside2uic import compileUi
    ModuleNotFoundError: No module named 'pyside2uic'

    ----------------------------------------
```

If I then manually install pyside2 using `pip3 install pyside2` and re-run `pip3 install joulescope-ui`, I get the following error:

```
Collecting joulescope-ui
  Using cached https://files.pythonhosted.org/packages/c5/cb/09bd4abd9645199378453501f3d18b955329199f9729880c0f18624e219b/joulescope_ui-0.4.5.tar.gz
ERROR: Packages installed from PyPI cannot depend on packages which are not also hosted on PyPI.
joulescope-ui depends on pyqtgraph@ https://github.com/jetperch/pyqtgraph/tarball/c8548b3246d29ee84a1ef76ebf63a5bb0e39c917#egg=pyqtgraph-0.11.0.dev0requests>=2.0.0
```

I'm using pip 19.1.1 with python 3.6.

mliberty1, 2019-07-08:
Hi @jeremyherbert and thank you for the detailed issue report.  I am able to duplicate this exact behavior on my macOS 10.14.5 2018 MacBook Pro (Python 3.7.2, pip 19.1.1) and also under Windows 10 x64 (Python 3.7.3, pip 19.1.1).  I am investigating how to best address this issue.  In the meantime, you can try cloning this repo and running from the source.  This works for me:

```
git clone https://github.com/jetperch/pyjoulescope_ui.git
cd pyjoulescope_ui
pip3 install -r requirements.txt
python3 setup.py qt
python3 -m joulescope_ui
```

mliberty1, 2019-07-08:
The first pyside2 issue is now fixed.  

The pyqtgraph dependency has no simple solution since this Joulescope UI package depends upon a forked version of pyqtgraph, at least until the pyqtgraph team releases 0.11.0.  I know that the pyqtgraph maintainers are actively preparing for this release, and I am hoping it's soon.  Rather than go outside pypi or push my own fork to pypi, I I have updated the README.md to recommend using the source code directly for the time being.  The change is the in the develop branch, and will be merged into master with the next release.

closed 2019-07-08 by mliberty1