jetperch/pyjoulescope_ui #6: [Feature Request] Export raw data

heroichornet, 2019-07-11:
Maybe it is already possible, but I have not found out how. Is there a simple way of dumping the measurements using the gui? Using the cli? Or is the .jls format easily convertible / readable?

mliberty1, 2019-07-11:
Yes and Yes!

To export data from the UI, capture the data of interest.  If this is less than 30 seconds, you can just pause streaming once the event occurs.  You can also record to a file, then load the file, and navigate to the region of interest.  Right click on the x-axis, click **Annotations**, click **Dual Markers**.  Left click on the left marker, move it to start of the data to export, left click again.  Left click on right marker, move it to the end of the data to export, left click again.  Finally, right click on either marker, click **Export data**, and click **Choose** for Filename,  You can select jls, csv or bin.  

You can also convert a jls file.  Check out:

    python -m joulescope recording --help

I should add bin, too.

heroichornet, 2019-07-12:
Thank you for the info, that's great news!

closed 2019-07-12 by heroichornet

mliberty1, 2019-07-12:
I should also mention that jls files are currently quantized to 0.2 seconds multiples.  If you record or attempt to save 1.19 seconds, the jls file will only contain 1.00 seconds of data.  This defect is on the list to fix.

wilda17, 2023-06-21:
is this still possible? I wasn't able to select anything else than *.jls in the export window

mliberty1, 2023-06-21:
Hi @wilda17 

Joulescope UI 1.0.22 only supports exporting to JLS v2.

This feature to export directly to other formats seems to have created more problems than it solved.  Many people tried to open long CSV exports in Excel, which only supports 1048576 rows, about 1 second of JS220 data.  Additional new features, such as multiple Joulescope support, further complicate these exports.

We do have it on our list to architect a new export to other formats feature.  If we can figure out a way to architect this without creating more problems, it will get added.  This feature is not high on our priority list.

However, using python, you have a lot of flexibility today.  You can capture directly to other formats with some work.  You can read JLS v2 files using python for analysis or format conversion.  For example, you can export the **current** signal:

    python -m pyjls export infile.jls --signal current outfile.csv

I am happy to help figure out how to get what you need.  What are you trying to accomplish?