jetperch/pyjoulescope_ui #7: Power consumption prediction

blackrosezy, 2019-07-15:
Can we have an option/view that can predict how long the device that is being test can be operate?

Example: A user enter 2500 mAh (a single AA battery), and the software will predict how long the device will be last (hours/days) base on the inputs from JouleScope.

I think this will simplify the process of calculation for the developer/engineer to design low power device. Thanks!

markcampanelli, 2019-07-15:
This sounds like something that might be better suited for a plug-in style Python extension to the basic Joulescope software. Has @jetperch considered adding a plugin capability?

mliberty1, 2019-07-15:
Thank you @blackrosezy for the suggestion.  Battery life estimation is definitely interesting, and it is on the "[Future Features](https://github.com/jetperch/pyjoulescope_ui/blob/master/features_future.md#other)" list under "other".  Battery life estimation depends upon battery chemistry, rate of discharge and your target (LDO vs switched mode), so it would take some time to implement correctly.  Do you have a simpler battery life model in mind?

Hi @markcampanelli, Yes, we have considered adding plugin capability to the Joulescope application, but I see it's not on the future features list yet.  I will add it.  The main holdup is that we want the plug-in API to be reasonably stable, the UI is not quite there yet.  For now, you can create Python applications using the joulescope package.