jetperch/pyjoulescope_ui #13: Prepackaged distribution fails to run on Linux Mint 18.1 (Ubuntu 16.04.x LTS)

ristomatti, 2019-07-24:
The prepackaged version seems to be compiled against libc6 2.25 but on my system only version 2.23 is available.

```
$ ./joulescope_launcher 
[19895] Error loading Python lib '/home/ristomatti/Tools/joulescope_0_4_6/libpython3.6m.so.1.0': dlopen: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.25' not found (required by /home/ristomatti/Tools/joulescope_0_4_6/libpython3.6m.so.1.0)
$ apt show libc6
Package: libc6
Version: 2.23-0ubuntu11
```

Also running `pip3 install -r requirements.txt` fails but I guess that is to be expected since the system provides only python 3.5. I was able to run it from source after installing python 3.7.0 using `pyenv` though, so `libc6` version didn't seem to matter.

mliberty1, 2019-07-24:
Hi @ristomatti - Great to hear you got the Joulescope software running under Linux Mint 18.1.  The downloadable, prepackaged Joulescope UI is compiled under Ubuntu 18.04 LTS.  For any other linux version, you will have to run from the python package, run from source or build the Joulescope UI on your computer. 

The Joulescope software does require Python 3.6 or newer.  

Eventually, we could get Joulescope picked up by the linux distro maintainers, but until then the downloadable linux version will be for just the latest Ubuntu LTS version.

closed 2019-07-24 by mliberty1

ristomatti, 2019-07-24:
@mliberty1 It's a fair requirement. I suggest updating the documentation at https://www.joulescope.com/download/ to reflect Ubuntu 18.04 LTS being a requirement. Currently the wording is "tested only on".

Oh well, I guess this is the signal for me to go and buy a backup USB drive, backup the system and try my luck with OS upgrade. Thanks for the quick reply!

mliberty1, 2019-07-24:
Running from python 3.7 and source or the pypi package through pip is perfectly fine, as you have confirmed.  Here is a relevant forum post:
https://forum.joulescope.com/t/installation-problems-on-ubuntu-16-04-lts/39

Now, if you want to upgrade your system, you certainly can.  However, if the Joulescope software launches and displays live data from your Joulescope, it will likely work just fine.

ristomatti, 2019-07-25:
Thanks for the extra information, it might be useful as a reference to someone else finding this issue. In my case I decided to take a leap of faith and upgraded my system in two phases last night, first from Mint 18.1 to 18.3 and then to 19.0 which is based on Ubuntu 18.04 LTS. Miraculously everything went without issues and now I'm running a newer OS and have easier access to various other packages also. After that I was able to run the prepackaged version just fine. :)