jetperch/pyjoulescope_ui #14: [Feature Request] Plugin architecture

mliberty1, 2019-07-29:
The Joulescope UI should have a plugin architecture so that we can easily add new features.

mliberty1, 2019-07-29:
I pushed up new code with which supports adding plugins to the dual markers in the "feature/plugins" branch.  For now, you will have to add your plugin to the "joulescope_ui/plugins" directory.  Then modify the joulescope_ui/plugins/__init__.py to import your plugin module and include it the PLUGINS_BUILTIN list.

The existing examples include:

- [joulescope_ui/plugins/example_01.py](https://github.com/jetperch/pyjoulescope_ui/blob/feature/plugins/joulescope_ui/plugins/example_01.py)
- joulescope_ui/plugins/usb_inrush.py
- joulescope_ui/exporter.py (not quite a plugin, but close)

See the [Plugin API](https://github.com/jetperch/pyjoulescope_ui/blob/feature/plugins/joulescope_ui/plugins/api.py) module for slightly more documentation.

markcampanelli, 2019-07-30:
Hi @mliberty1. Thanks for taking pn this useful feature. Python 3.7 offers a way to do fairly lightweight plugin discovery through dynamic module attributes. See https://realpython.com/python37-new-features/#customization-of-module-attributes.

mliberty1, 2019-07-30:
Thanks for the pointer @markcampanelli !  The work so far has focused more on the API back to Joulescope from the plugin rather than discovery, since I have someone who needs this now.  I will be figuring out how to make both builtin (distributed with the UI) and extension (code place in some specific plugin directory) work more seamlessly.  Recommendations, opinions and references to python code with exceptional plugin systems welcome!

markcampanelli, 2019-07-30:
Here's an alternative discovery option that seems to be geared more towards inter-library plugin discovery: https://blog.danallan.com/posts/2019-07-24-use-entrypoints-more/

mliberty1, 2024-04-30:
Available in 1.1.6.

closed 2024-04-30 by mliberty1