jetperch/pyjoulescope_ui #15: [Feature Request] Time Information for Statistics

tomisaarnio, 2019-08-05:
Joulescope 0.5.0 running on 64-bit Windows 10.
 
Please consider adding time information for the following data:

1. Oscilloscope view, Energy
- The cumulative energy in the Control-pane should have elapsed time next to it in order to help estimating device energy consumption and show how long energy measurement has been running since last Clear energy.
- The elapsed time should be updated as cumulative energy is updated in order to keep them in sync.
- The elapsed time should be reset together with cumulative energy when user selects Tools | Clear energy.
- Bonus: It would also be very helpful to show also cumulative mAh along with the cumulative energy and elapsed time. This would help with battery capacity estimation.

2. Oscilloscope view, Trace statistics
- The oscilloscope trace statistics for each trace should also display time span for the data. Now it is a bit difficult to determine what is the time span of the statistics calculated.

3. Multimeter view
- Multimeter view should also show elapsed time. Especially during accumulation, the elapsed time is essential.

mliberty1, 2019-08-05:
Understood and agreed.  I think that going forward the request is to show time duration wherever the UI displays accumulated/integrated quantities.  The 0.5.0 only does this clearly for dual markers.  The quantities to right of waveforms and on the multimeter, single value, and control views do not.  The 0.5.0 software only displays charge as C (coulombs), not mAh, and should be able to convert for you.

mliberty1, 2019-12-04:
Items (2) and (3) are addressed in 0.7.0.  Item (1) is still not implemented.

closed 2019-12-04 by mliberty1