jetperch/pyjoulescope_ui #19: [Feature Request] Time delta numbers only show in engineering notation

efox29, 2019-09-03:
When zooming into a waveform and measuring between two points, delta is represented in scientific notation. It would be beneficial to use engineering notation since that is the notation that we use. 

[image]

mliberty1, 2019-09-03:
Agreed.  This feature is loosely related to [this request](https://forum.joulescope.com/t/x-axis-division-labels/126) from the Joulescope forum.

mliberty1, 2019-09-04:
Fixed in 2a43a82 which will be included in the next release:
[image]

closed 2019-09-04 by mliberty1