jetperch/pyjoulescope_ui #27: Wishlist - automatic power on

jamesl-dm, 2019-11-01:
I keep my joulescope permanently plugged into my power supply, so having the output disabled by default at startup can get a bit annoying. Every morning I plug a new prototype or candidate for repair into the supply, and wonder why it isn't lighting up. This is before the coffee hits, so it takes a moment to realize that I need to open the software to get it to turn on.

Presumably this is for safety... But it would be great if there was a non-volatile firmware setting that would make it turn on by default.

mliberty1, 2019-11-01:
Joulescope is designed to disconnect IN+ from OUT+ whenever its isolated sensor side is unpowered.  Turning on Joulescope's sensor side requires more current, which cannot happen until the device enumerates over USB into high-speed mode.  Windows fully enumerates devices automatically, and we could support this feature.  Linux and macOS defer until software requests it, so we can't automatically turn on the Joulescope sensor at host computer boot.

What OS are you using?

If the distributed Joulescope UI software had a command-line option to turn on the sensor for all connected Joulescopes and configure them to operate normally, would that work for you?  You could configure you host computer to [call this program at startup](https://www.howtogeek.com/228467/how-to-make-a-program-run-at-startup-on-any-computer/).

jamesl-dm, 2019-11-04:
I'm on windows - but command-line option sounds like a good plan.

mliberty1, 2020-03-16:
With the next release of the Joulescope UI executable, you connect IN+ to OUT+ on all Joulescopes attached to the host computer like this:

    C:\Program Files (x86)\Joulescope\joulescope.exe parameter_set --config auto

The command also allows you to specify a Joulescope by serial number, and also set other parameters.  See the help for details:

    joulescope.exe parameter_set --help

If you have python installed with the joulescope package, you can use the joulescope.exe executable which is installed as part of the package.  You can also use the package itself:

    python3 -m joulescope parameter_set --help

jamesl-dm, 2020-03-17:
Thanks Matt!

closed 2020-03-17 by jamesl-dm