jetperch/pyjoulescope_ui #28: Wishlist - toolbar button for toggling the min/max lines

jamesl-dm, 2019-11-01:
One of the most frequent operations for me is turning the min/max lines off as required. They are super-handy when you need to see the peaks, and super irritating when you don't. It would be nice if the toggle for them wasn't buried in the settings menu - maybe a hotkey or toolbar button?

mliberty1, 2019-11-01:
Noted!  We are currently architecting a change to how the Joulescope UI manages preferences.  We will add this feature, likely as a right-click context menu option.

mliberty1, 2019-12-04:
Added "Waveform Control" widget to 0.7.0, which includes Min/Max lines as shown here:
[image]

closed 2019-12-04 by mliberty1