jetperch/pyjoulescope_ui #29: Frequent crashes when removing markers

jamesl-dm, 2019-11-01:
I don't have a reliable recipe for replicating this, but in 0.6.8 I get frequent crashes when removing markers, if I have stopped and started the capture in-between marker operations and zoom changes. Usually with multiple marker pairs active.

Disclaimer - entirely possible its a graphics driver issue on my end.

mliberty1, 2019-11-01:
What OS are you running on the host computer?  Are you running the official Joulescope UI binary distribution?  If you are running from source, could you confirm that you are using our patched version of pyqtgraph?

jamesl-dm, 2019-11-04:
Hi Matt - its the windows official binary. It's quite tricky to reproduce - I've tried a bunch of times, and can't figure out what combination of zooms, restarts, and marker positions does it. But if you copy the attached video exactly, you should be able to reproduce it, assuming the issue isn't on my end.

[2019-11-04 10-25-09.zip](https://github.com/jetperch/pyjoulescope_ui/files/3802533/2019-11-04.10-25-09.zip)

mliberty1, 2019-11-04:
Hi @jamesl-dm - I've tried repeating the actions in the video as closely as possible, and I have not yet been able to duplicate the locked-up behavior you recorded.  However, I do see a difference.  At 0:15 in your video, the statistics for the marker at 26.77 disappear.  On my Win 10 PC with 0.6.8, the marker continues to display the old statistics values.  If the software inconsistently handles this text, it could be causing the lock-up behavior you see.  We will continue to investigate.

Is it possible for you to send me the Joulescope logs from your PC?  They are located in:

    C:\Users\{your_user_name}\AppData\Local\joulescope\log

For example, for me with username mliberty, it is:

    C:\Users\mliberty\AppData\Local\joulescope\log

The logs filename is the date and time in UTC.  You just need to send the log from the crash, but if you don't know which one that is, feel free to send a bunch and I will inspect to figure it out.

jamesl-dm, 2019-11-05:
Here's the log - sadly nothing of interest at the end:
[joulescope_20191104_022504_13024.log](https://github.com/jetperch/pyjoulescope_ui/files/3806895/joulescope_20191104_022504_13024.log)

Maybe using Windows 8 has something to do with it... If there's no obvious answer, I'll build from source and try to debug on this end.

mliberty1, 2019-11-05:
Hi @jamesl-dm - I had another person report a similar issue with markers and Joulescope UI freezes last night using 0.6.8 on Win 10 x64.  I don't think it is just you.  We will also be investigating.

mliberty1, 2019-11-11:
Hi @jamesl-dm - We have been making progress, and have identified two separate issues that can lead to this behavior:

First, markers can be referenced after they are deleted.  The marker implementation ends up being complicated.  Markers are related to the waveform, the x-axis, each signal, and the underlying data model in streaming, buffer and file modes, each of which has slightly different performance optimizations.  To complicate things further, the dual marker values must be separately computed asynchronously.  We have too many places that hold on to marker instances (and the text statistics related to each signal), especially when they can be deleted at any time.  The right answer is to contain the marker objects and instead reference them by string name.  We are making this change as part of our "Preferences" refactoring, which is really a major restructuring of the Joulescope UI application.  

Second, the underlying pyqtgraph implementation seems to be very touchy when removing objects from the Qt scenegraph.  We will investigate and make sure we do this correctly.

jamesl-dm, 2019-11-12:
Hi Matt - thanks for the update. Since it sounds like the work is in the pipeline already, I'll close the issue.

closed 2019-11-12 by jamesl-dm