jetperch/pyjoulescope_ui #55: Add option to display waveform time as UTC

mliberty1, 2020-03-11:
Versions: 0.8.6
Platforms: All

The Joulescope time is currently displayed as seconds relative to the buffer, including when loading files.  For synchronizing to other captures, including UART, would be good to be able to display time as UTC.

mliberty1, 2020-03-13:
Sometimes, manually setting a "time zero" is also useful.  For example, when synchronizing to other equipment using a GPI trigger.  The Joulescope UI should allow the user to specify time zero and adjust the x-axis time labels accordingly.

mliberty1, 2020-03-24:
Also show calendar time on markers:
See https://forum.joulescope.com/t/actual-time-display-when-dragging-marker/220

mliberty1, 2020-07-09:
Also include the option to format the date & time.  Options include:

* [ISO 8601](https://en.wikipedia.org/wiki/ISO_8601)
* YYYY-MM-DD hh:mm:ss.nnnnnn
* MM-DD-YYYY 
* DD-MM-YYYY
* Timezones: local or UTC
* am/pm or 24 hour

gideon-decr, 2020-10-20:
+1 on this request

fnaeem786, 2022-01-31:
+1 on this
Do you have an ETA?

mliberty1, 2023-03-10:
We have finally implemented this feature just under three years to the day.  The x-axis is now UTC time, and recording to JLS v2 files also saves the UTC time.  When you open the JLS v2 file, it displays the correct UTC time.

closed 2023-03-10 by mliberty1