jetperch/pyjoulescope_ui #72: UI should have a "dark" theme option

mliberty1, 2020-04-11:
Version: 0.8.11
Platforms: All

**Description**
The existing UI uses the default window colors for each platform.  The preferences allow some color settings, but the flexibility is limited to certain Widgets.

**Desired Behavior**
The UI should contain a preference that would set the existing default theme or dark theme for all widgets.

mliberty1, 2020-07-01:
Hi @Ivanovitch_k

Just checking to see if you looked into this at all.  If I remember correctly, you expressed interest in implementing the dark theme for the Joulescope UI.

I came across this:
https://stackoverflow.com/questions/48256772/dark-theme-for-qt-widgets

The leading answer looks great, and is MIT licensed:
https://github.com/Alexhuszagh/BreezeStyleSheets

I would like to try this out for the upcoming release.  Are you still interested in doing this?  If you haven't started and don't have time in the next few days, I will go ahead and take a look.

Ivanovitchk, 2020-07-02:
Hi @mliberty1 

I had some local tweaks running but couldn't allocate enough time on it lately,

Breeze Stylesheet looks quite promising & clean looking. Go head, I'll see If I can chime in !

mliberty1, 2020-08-12:
Added in 0.9.1.

closed 2020-08-12 by mliberty1