jetperch/pyjoulescope_ui #79: Provide Flatpak

dbrgn, 2020-06-08:
Flatpak (https://www.flatpak.org/) is a packaging standard for desktop apps that can be used with any Linux distribution. Some distributions like Fedora (and Ubuntu?) support Flatpak out of the box.

Would it be an option to package the Joulescope software as a Flatpak package and distribute it via Flathub? That would make installing much easier.

(An alternative would be https://appimage.org/, but Flatpak probably offers the better system integration.)

mliberty1, 2020-06-09:
This reminds me of [XKCD 927](https://xkcd.com/927/).  apt, snap, pacman, dnf, pip, ...

Are you experienced deploying with Flatpak?  Would you (or someone you know) be interested in creating the Flatpak configuration for the Joulescope UI?  It is probably quick for someone who has done it before...

dbrgn, 2020-06-09:
I haven't done it before, but would be willing to contribute an initial configuration as long as someone would take over from that for maintenance :slightly_smiling_face: 

In contrast to other package managers like apt or pacman, flatpaks are not distro-specific. Additionally, multiple runtimes and package versions can be installed in parallel. On top of that you get sandboxing and a permission model. It's quite an interesting project.

dbrgn, 2020-06-17:
Hm, I gave it a try, but it seems that building Python packages with many transitive dependencies is really tedious, and that tooling (https://github.com/flatpak/flatpak-builder-tools/tree/master/pip) is not really mature yet (e.g. it cannot deal with comments or version ranges in requirements.txt).

Maybe this should be revisited in 1-2 years.

mliberty1, 2020-06-17:
Thanks for investigating @dbrgn .  Sounds like a good technology to monitor for the future.  For now, I will go ahead and close this issue.  Feel free to reopen it or a new one when you feel the time is right.

closed 2020-06-17 by mliberty1

dbrgn, 2020-06-17:
Yep, that's a good idea. Flatpak is really cool for C/C++ applications with little dependencies (check out for example how easy packaging [LibrePCB](https://github.com/flathub/org.librepcb.LibrePCB/blob/master/org.librepcb.LibrePCB.json) is), but packaging Python with a lot of dependencies is a bit tedious without good tooling.

Another distro-independent binary format would be [AppImage](https://appimage.org/). It provides double-clickable applications that "just run" on any Linux distribution. Python packaging [is a bit involved though](https://github.com/AppImage/AppImageKit/wiki/Bundling-Python-apps). I guess the reason is that Python packaging in general sucks, especially with native dependencies like Qt :confused:

mliberty1, 2020-06-17:
Interestingly, LibrePCB does not include its dependencies.  The project page says that it requires:

- g++ >= 4.8, MinGW >= 4.8, or Clang >= 3.3 (C++11 support is required)
- Qt >= 5.2
- zlib
- OpenSSL

I would think that these need to show up somewhere in Flatpack file, otherwise the user still has to install them in a platform-specific manner.  

Packaging is a major problem with all programming languages.  Languages/frameworks like node.js, Python and Elixir take this on separate from the platform, which means they get in each other's way, leading to things like venv and containers.

dbrgn, 2020-06-17:
That is all part of the runtime. Flatpak puts commonly used functionality in a so-called runtime, which is similar to a Docker layer or a distribution with base packages. In contrast to a distribution, where you can either run Ubuntu 16.04 or Ubuntu 18.04, you can have multiple runtimes and versions in parallel. A runtime guarantees a consistent and reproducible application context, no matter what distro you're using.

Here you can see what the runtimes entail: https://docs.flatpak.org/en/latest/available-runtimes.html LibrePCB includes the KDE runtime, that's why Qt libraries are preinstalled. And it uses the KDE SDK, which provides the C++ compiler at build time.

Flatpak is really interesting. It also puts [sandboxing](https://docs.flatpak.org/en/latest/sandbox-permissions-reference.html) on top (a bit similar to app sandboxing on mobile platforms). You can also request temporary permissions (e.g. camera or file system access) through so-called portals.