jetperch/pyjoulescope_ui #92: [feature req] Remembering last saved location

efox29, 2020-08-20:
When exporting the data, the default path My Documents/joulescope, but I may want to save to another location with multiple exports, and I would have to navigate to the path each time. 

Would be great if when saving, the last path is remembered.

mliberty1, 2020-08-20:
The 0.9.2 UI has a data_path preference that determines the path for all JLS loads & saves:
[image]

I think that this feature request is to add another preference, perhaps "data_path_type" as one of:
* "Most recently saved": update the load/save path whenever you save a JLS file 
* "Most recently used": update the load/save path whenever you load or save a JLS file
* "data_path": Always use the configured path.

What do you think?

mliberty1, 2020-11-25:
Added in 0.9.3.

closed 2020-11-25 by mliberty1