jetperch/pyjoulescope_ui #93: [feature req] Ability to add board and additional infomation (text annotation) to joulescope

efox29, 2020-08-20:
It would be great to have a "properties" or "meta" tab that would add additional information to each export or screenshot. 

I can imagine an area where the user can enter some text which is displayed on the screen but out of the way of the measurement information. When the user exports the data, that annotated text is embedded into the joulescope export. This will allow a user to enter in board information and description of tests or measurement setup and will enable them to conduct multiple tests and export various sets of data and share them with clarity. The same could apply to screenshots or image exports.

mliberty1, 2020-08-20:
I have a few questions:

I think that saving this additional text information to the JLS files should be straightforward.  When you load a JLS file, how should the UI let you see this information?

For the screenshot, do you intend this feature for only the waveform widget?  Do you mean the new "Save image" and "Copy image to clipboard" features on the waveform widget?  If so, the UI would need to have a way to display this text, and then you could "Save image".  Perhaps an optional new row below the last signal?

efox29, 2020-08-20:
> how should the UI let you see this information?

I could imagine similar to how the statistics are currently displayed on the right side of the screen. Perhaps display text on the right or a row at the top or bottom. It might make sense to limit the number of characters to some fixed amount.  The intent is to describe loosely where the data is from, or a brief description of what is happening and not go into full detail.

mliberty1, 2023-07-25:
Added to 1.0.27

closed 2023-07-25 by mliberty1