jetperch/pyjoulescope_ui #96: Unable to clone and run Joulescope UI on Ubuntu 18.04

nithinn86, 2020-09-18:
I downloaded joulescope UI version 0.9.2 and executed the following steps on my Ubuntu 18.04 machine as explained in the README:

```
pip3 uninstall pyqtgraph
cd {path/to/repos}
git clone https://github.com/jetperch/pyjoulescope_ui.git
cd pyjoulescope_ui
pip3 install -U -r requirements.txt
python3 setup.py qt
```

It throws the following error:

```
running qt
Traceback (most recent call last):
  File "setup.py", line 212, in <module>
    'Source': 'https://github.com/jetperch/pyjoulescope_ui/',
  File "/usr/local/lib/python3.6/dist-packages/setuptools/__init__.py", line 153, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib/python3.6/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "setup.py", line 117, in run
    convert_qt_ui()
  File "setup.py", line 50, in convert_qt_ui
    rcc_path = qt_rcc_path()
  File "setup.py", line 44, in qt_rcc_path
    raise ValueError('Could not find rcc executable')
ValueError: Could not find rcc executable
```

Please note that this worked fine for UI versions 0.8.xx.

mliberty1, 2020-09-18:
The Joulescope UI uses PySide2 to provide the bindings to Qt.  Unfortunately, PySide2 has had some stability issues recently, and we had to fix a specific version for python 3.6, which is used by Ubuntu 18.04 LTS.  Windows (the most popular OS for Joulescope UI) needed the latest PySide2, but Ubuntu 18.04 LTS needed to stay back.

To complicate things, PySide2 also broke (at least from our perspective) the resource packaging, so we have to find the actual executable (not the python wrapped executable) which is now included with PySide2.  It's a dependency versioning nightmare.  With all these problems combined with the majority of our users are Windows or Mac or have already migrated to Ubuntu 20.04 LTS, we made the decision to not support Ubuntu 18.04 LTS.  I apologize that this issue is affecting you.

I think that you may now be able to update to PySide2 5.15.1 on 18.04 LTS, but I am not sure.  I not longer have a system to quickly test this.  Here's how to try by running from source:

First, uninstall joulescope_ui and upgrade PySide2:

    cd {your repo path}
    pip3 uninstall joulescope_ui
    pip3 install -U PySide2

You should see that 5.15.1 gets installed.  Now, clone the Joulescope UI and build the Joulescope UI QT resources

    git clone https://github.com/jetperch/pyjoulescope_ui.git
    cd joulescope_ui
    python3 setup.py qt

And now run it:

    python3 -m joulescope_ui

If you want to install the joulescope UI, you can edit requirements.txt lines 14 through 16 to remove the existing PySide2 lines and replace them with:

    PySide2>=5.15.1

And then install the joulescope_ui:

    pip3 install .

You should then be able to run `python3 -m joulescope_ui` from any path.

curiousmuch, 2020-10-29:
Hey @mliberty1, 

I can confirm this fixed the issue for me.

mliberty1, 2020-10-29:
Great!  I will close the issue, but feel free to reopen it if you run into any more issues.

closed 2020-10-29 by mliberty1