jetperch/pyjoulescope_ui #100: Run in Raspberry pi

Moo2017, 2020-10-14:
Hi All,
I was wondering if it's possible to run this on raspberry pi board? I tried to install that but then I got some errors related to PisSide2. Also I installed Ubuntu in Ras
Any help would be highly appreciated.

Masi

mliberty1, 2020-10-14:
We have not tested Joulescope with a Raspberry Pi.  Joulescope really wants a 64-bit operating system.  Until the latest Raspberry Pi 4, all Raspberry Pis ran 32-bit Linux.  What Raspberry Pi version and RAM size are you trying to use?

Moo2017, 2020-10-15:
Thanks for the quick response. I am using "Raspberry Pi 4 computer Model B 4GB RAM".

mliberty1, 2020-10-15:
I just ordered a Raspberry Pi 4 B 8GB RAM, and it should arrive by Monday.  As long as you run 64-bit Linux with the right dependencies, it should work.  I will check it out, initially with Raspian, and let you know.

If you do not hear back from me by Oct 21, please comment here to ask me how things are going!

Moo2017, 2020-10-15:
Sounds great, thank you very much.

Moo2017, 2020-10-15:
@mliberty1 Also I wanted to let you know that I've installed Ubuntu 20 on my raspberry pi and then tried to install the joulescope and I've got this error after I ran "pip3 install -U -r requirements.txt" :

```
RROR: Could not find a version that satisfies the requirement PySide2>=5.15.0 (from -r requirements.txt (line 16)) (from versions: none)
ERROR: No matching distribution found for PySide2>=5.15.0 (from -r requirements.txt (line 16))
```

mliberty1, 2020-10-15:
Do you know what python version you are running?  Here's how to tell:

    python3 -VV

(that's two 'V', not 'W')

Moo2017, 2020-10-15:
it's 3.8.5

mliberty1, 2020-10-15:
I just came across [this](https://forum.qt.io/topic/112813/installing-pyside2-on-raspberry-pi).  Running a search on [Ubuntu packages](https://packages.ubuntu.com/search?keywords=pyside2&searchon=names&suite=focal&section=all) shows that you can install PySide2 5.**14**.0, but not PySide2 5.**15**.0 that the Joulescope UI currently uses.

We may need to build Qt + PySide2 from source...

Moo2017, 2020-10-15:
I only found this link "https://pyside.readthedocs.io/en/latest/building/linux.html" which is for Ubuntu 12- 14 , not sure if it works on Ubuntu 20. Any idea how we should build Qt + Pyside2?

mliberty1, 2020-10-15:
Not yet.  I will have to go through the process once I get my Raspberry Pi 4, and I'll write it up.  Let me know if you beat me to it!

Moo2017, 2020-10-15:
Thanks! I don't think I can beat you!, however, let you know if any magic happens!

Moo2017, 2020-10-15:
what if I install Ubuntu 18 since there is a built version for Ubuntu 18.04?

mliberty1, 2020-10-15:
Ubuntu 18.04 LTS only comes with python 3.6, which I think also has some PySide2 dependency issues.  It may work though, but you will have to remove the PySide2 version from requirements.txt.

mliberty1, 2020-10-20:
I spent a few hours struggling to get the Raspberry Pi build working.  It's not.

Here's a [gist](https://gist.github.com/mliberty1/4bc1ed6ec23904e5be66fda54f9ad75d) of what I tried:

Moo2017, 2020-10-22:
@mliberty1 Thanks so didn't work at the end right? Maybe I can buy a mini PC and install windows 10 instead of raspberry pi, what do you think?

mliberty1, 2020-10-22:
It is not working yet.  I ran out of time, but I plan to spend more time next week.  The failure is with configuring the build process correctly, which I should be able to solve.  I will likely run into other build problems, though.

The challenge is that Ubuntu does not have PySide2 packages for arm64.  If you just want to capture data, not using the Joulescope UI, you can use the joulescope command-line tools.

You can run the Joulescope UI on Ubuntu on any x86 64-bit architecture.  I really like Intel NUCs.

Moo2017, 2020-10-22:
What I need is to connect the Joulescope to my device and set it to capture the current consumption of my device every n second and store the result for e.g. a day and then get the result. I don't need UI just I want to be able to configure the current reading interval and also to log the data. I can later write an script to create the current consumption curve based on the stored data. Could you please let me if this is possible? Where I can find the joulescope command-line tools?

mliberty1, 2020-10-22:
Ok, running from the command line is much easier!  I recommend you use [downsample_logging.py](https://github.com/jetperch/pyjoulescope_examples/blob/master/bin/downsample_logging.py).  Here are the instructions:

# Install dependencies

    sudo apt update
    sudo apt upgrade
    sudo apt install git python3-pip python3-venv

# Configure Joulescope permissions

    wget https://raw.githubusercontent.com/jetperch/pyjoulescope/master/99-joulescope.rules
    sudo cp 99-joulescope.rules /etc/udev/rules.d/
    sudo udevadm control --reload-rules

Connect your Joulescope

# Configure Joulescope software in virtual environment

    python3 -m venv ~/venv/joulescope
    . ~/venv/joulescope/bin/activate
    pip3 install -U joulescope
    joulescope info

You should see something like:

    System information
        Python: 3.8.5 (default, Jul 28 2020, 12:59:40)
    [GCC 9.3.0]
        Platform: Linux-5.4.0-1022-raspi-aarch64-with-glibc2.29 (linux)
        Processor: aarch64
        executable: /home/ubuntu/venv/joulescope/bin/python3
        frozen: False
    
    joulescope version: 0.9.1
    Found 1 connected Joulescope:
        Joulescope:000416   ctl=1.3.2            sensor=

# Get downsample logging

    mkdir ~/repos && cd $_
    git clone https://github.com/jetperch/pyjoulescope_examples.git

# Run

    cd pyjoulescope_examples
    python3 bin/downsample_logging.py --help
    python3 bin/downsample_logging.py --source sensor --downsample 2

The "--source sensor" option uses the new on-instruments statistics computation.  Without this option, your Joulescope sends all 2 Msps data to the Raspberry Pi, and the Raspberry Pi computes the statistics.  With this option, the statistics are computed on your Joulescope, and only the statistics are sent to the Raspberry Pi for much lower data rate.

The collected data is recorded to ~/Documents/joulescope.

# Run after reboot

Once you have installed all of the above, running again is much quicker:

    . ~/venv/joulescope/bin/activate
    python3 ~/repos/pyjoulescope_examples/bin/downsample_logging.py --source sensor --downsample 2

Moo2017, 2020-10-22:
Thank you very much, I'll git it a try and let you know if there is any issue

Moo2017, 2020-11-06:
Thanks @mliberty1 , it worked fine. Could you please also let me know how I can change the sampling period? I am assuming that the Current is the average current during the sampling period( which in the above example is 1 second), is that right?

mliberty1, 2020-11-06:
Great!  Yes, the current, voltage, and power are all the average over the sampling period.  Charge and energy are integrated over the duration of the program.

Moo2017, 2020-11-06:
Thanks. We would like to run the joulescope to log the data every day and store it in a different file (I mean for each day, stores the data in one file), is there any example to do this?

mliberty1, 2020-11-06:
We do not have any example that switches log files like this.  The [statistics.py](https://github.com/jetperch/pyjoulescope_examples/blob/master/bin/statistics.py) example is a decent, simple starting point.  However, [downsample_logging.py](https://github.com/jetperch/pyjoulescope_examples/blob/master/bin/downsample_logging.py) is built to be more robust than statistics.py.  If you want to try for downsample_logging, you just need to close and then open new files daily.  Take a look at [self._f_csv](https://github.com/jetperch/pyjoulescope_examples/blob/master/bin/downsample_logging.py#L446).  Getting the resume feature to work with downsample_logging could be more challenging though.

mliberty1, 2020-11-13:
Running pyjoulescope on the Raspberry Pi with 64-bit Raspian or Ubuntu works great.  Running pyjoulescope_ui does not yet work due to insufficient PySide2 support.  I am going to close this issue since it seems like far too much work at this point in time, and you do not need this feature.  Feel free to reopen it if running the UI becomes important to you.

closed 2020-11-13 by mliberty1

DronXa, 2022-05-21:
Is it possible to get the UI working in raspberry pi, I actually need this feature for a current project of mine. 

Thanks for the help!

mliberty1, 2022-05-21:
Hi @DronXa - The problem is not the Joulescope UI code itself, but the graphics library dependency on PySide2 / PySide6 & Qt.  The OS distributions did not include prebuilt packages, at least when I last looked.  [This recent post](https://forum.qt.io/topic/135892/installing-pyside-on-a-raspberry-pi) is not encouraging.  You may be able to build from source.  However, it's probably easier to get an inexpensive [Intel NUC](https://www.intel.com/content/www/us/en/products/details/nuc.html) or equivalent.

Our official UI distribution is still on PySide2, but you can run PySide6 with the feature/pyside6 branch.

DronXa, 2022-05-21:
> Hi @DronXa - The problem is not the Joulescope UI code itself, but the graphics library dependency on PySide2 / PySide6 & Qt. The OS distributions did not include prebuilt packages, at least when I [...]

Thanks for the respons, the main reason as to why I need to run Joulescope on a raspberry pi, is to be able to monitor a DUT in  an enclosed EMC chamber. The idea was to connect to the raspberry pi via an ethernet cable and monitor the DUT in GUI via VNC. I just saw pyjoulescope_server repo and I wonder if it is possible to use joulescope_server on a raspberry pi  and just feed the data live to a different client that can run Joulescope UI.

mliberty1, 2022-05-21:
> I wonder if it is possible to use joulescope_server on a raspberry pi and just feed the data live to a different client that can run Joulescope UI.

You can use joulescope_server or just pyjoulescope, but the Joulescope UI does not currently support connecting to a joulescope_server.  If you just need statistics updates at 2 Hz, you can ssh to the Raspberry Pi to run [`statistics.py`](https://github.com/jetperch/pyjoulescope_examples/blob/master/bin/statistics.py) or [`statistics_logger.py`](https://github.com/jetperch/pyjoulescope_examples/blob/master/bin/statistics_logger.py).

If you need full-rate data, I would not recommend a Raspberry Pi.  EMC testing is expensive.  An Intel NUC or inexpensive laptop is pretty affordable.

mliberty1, 2024-04-25:
Qt6 now comes prebuilt for arm64 / aarch64, so I decided to try again.  Unfortunately, still no luck.  PySide6-QtAds does not prebuild with arm64, and I ran into issues building it.  GitHub Actions will be adding Linux arm64 this year (in private beta now).  Eventually, we can get the PySide6-QtAds project to provide the wheel.

I actually tried on both Raspberry Pi OS and Ubuntu 23.10.  The results where the same.  Here are the steps:

```
Raspberry Pi 5 (8 GB RAM)
Raspberry Pi OS 64-bit

sudo apt update
sudo apt upgrade
sudo apt install build-essential python3.11-dev python3.11-venv git libudev-dev  cmake

# Create Python virtual environment
python3 -m venv ~/venv/joulescope
. ~/venv/joulescope/bin/activate
pip install pyjoulescope_driver
pip install shiboken6 PySide6

mkdir ~/tmp && cd ~/tmp

# Checkout PySide6-QtAds
# See https://github.com/mborgerson/pyside6_qtads/ for latest build instructions
git clone https://github.com/mborgerson/pyside6_qtads.git
cd pyside6_qtads
git submodule init
git submodule update

# Download Qt
QT_BASE_DIR=/home/joulescope/tmp/qt
QT6_DIR=${QT_BASE_DIR}/6.7.0/gcc_arm64
pip install aqtinstall
aqt install-qt linux_arm64 desktop 6.7.0 --outputdir ${QT_BASE_DIR}

# Build PySide6-QtAds
sudo apt install libxkbcommon-dev
pip install cmake-build-extension setuptools wheel shiboken6-generator
LD_LIBRARY_PATH=${QT6_DIR}/lib \
CMAKE_PREFIX_PATH=${QT6_DIR}/lib/cmake/ \
PIP_EXTRA_INDEX_URL=https://download.qt.io/official_releases/QtForPython/ \
VERBOSE=1 \
python setup.py bdist_wheel
```

BUT get:

```
/home/joulescope/venv/joulescope/lib/python3.11/site-packages/shiboken6_generator/shiboken6: error while loading shared libraries: libLLVM-11.so.1: cannot open shared object file: No such file or directory

ls /home/joulescope/venv/joulescope/lib/python3.11/site-packages/shiboken6_generator/
has libclang-11.so.1, but no libLLVM-11.so.1
```

I could not find libLLVM-11.so.1 for arm64, and I failed to build from source.

I also tried to build shiboken6-generator:

```
# https://github.com/qtproject/pyside-pyside-setup/blob/dev/README.md
sudo apt install clang
wget https://github.com/qtproject/pyside-pyside-setup/archive/refs/tags/v6.7.0.zip
unzip v6.7.0.zip
cd pyside-pyside-setup-6.7.0
python setup.py build --build-type=shiboken6-generator --qtpaths=${QT6_DIR}/bin/qtpaths6

which results in yet another problem:
CMake Error at cmake/ShibokenHelpers.cmake:194 (_message):
Call Stack (most recent call first):
  /usr/lib/llvm-14/lib/cmake/clang/ClangTargets.cmake:756 (message)
  /usr/lib/cmake/clang-14/ClangConfig.cmake:19 (include)
  cmake/ShibokenHelpers.cmake:170 (find_package)
  cmake/ShibokenSetup.cmake:37 (setup_clang)
  CMakeLists.txt:13 (include)
```

If this had worked, the remaining steps should be:

```
# Configure for Joulescope
pip install joulescope_ui
wget https://raw.githubusercontent.com/jetperch/joulescope_driver/main/99-joulescope.rules
sudo cp 99-joulescope.rules /etc/udev/rules.d/
sudo udevadm control --reload-rules
```