jetperch/pyjoulescope_ui #101: Horizontal cursor

efox29, 2020-10-15:
It might be nice to have a horizontal cursor such that the user can specify a visual aid for a threshold. Sometimes I am looking for the current to exceed a certain value. It would be nice to visually see all those points without zooming into the details (since triggering on current is not possible to my knowledge).

mliberty1, 2020-10-15:
Hi @efox29.  Before I start asking questions, perhaps you have not yet come across horizontal markers:
[image]

This feature is now in the most recent 0.9.2 release, and it is admittedly not the most easily discoverable feature.  To add a horizontal marker, right-click (control-click on mac) on the y-axis, like this:
[image]

Does this already do what you want?

efox29, 2020-10-15:
Derp.

This is exactly it, I saw that menu, but when I saw single and dual markers, I just assumed, it was for the vertical (which is what I use most of the time).

Closed.

closed 2020-10-15 by efox29