jetperch/pyjoulescope_ui #102: Optimize JLS responsiveness for very large files

mliberty1, 2020-10-19:
Version: 0.9.2
Platform: All

**Description**
One common Joulescope use case it to log data for very long durations.  Loading 2.75 days of 10 kHz data (36 GB) takes 2 minutes, 30 seconds.  Although 36 GB is a lot of data, faster load and access times would be appreciated.

**Expected Behavior**
Ideally, all data accesses would be fast.

**Developer Notes**
The JLS file read code is currently written in Python, and likely has a lot of room for improvement.  The code also only maintains a single index.  For long files, it could maintain multiple indices for faster direct access.

gideon-decr, 2020-10-20:
If you want to improve your file size and read times then you need to reduce entropy in the signal, compress, AND exploit special processor features.
This package helps you quickly play with those features:
https://github.com/searchivarius/PyFastPFor
https://github.com/searchivarius/PyFastPFor/blob/master/python_bindings/examples.ipynb

Entropy:
The simplest way of reducing entropy in time series signals is differencing:
delta1 
delta4
(Differencing works because in real world signals neighboring values lie close to one another.)

Special processor features:
SIMD based packing. Check out the codecs like:
simdfastpfor128

If you want to get WAY more sophisticated then you can start looking at decomposing the signal into different spaces that reduce entropy even more. You should probably bring in a DSP expert if you want to start doing that type of game. You want to look at typical signals that people record (start asking for signal donations now so you can build up a representative data set) and then build up the ability to analyze the stored signal then chose how to decompose it and then go back over the signal after it is finally stored and recompress it once you know how it looks.

mliberty1, 2020-10-20:
@gideon-decr Thanks for the pointer to PyFastPFor.  I was not familiar with that one.  I did look into a bunch of compression alternatives at one point, both lossless and lossy, but JLS files are still uncompressed.

Unfortunately, the existing software is not strictly IO-bound.  I was able to read the 36 GB file in 30.3 seconds using 1 MB chunks in python.  It takes the UI 150 seconds (5x longer) to display the waveform.

I did some tests, and the problem is with our storage of downsampled data.  Loading a full-rate 2 MSPS 23 GB file (2941 seconds) only took 2.71 seconds.  If we extrapolate, a 36 GB file should take 4.24 seconds.  Loading the same sized downsampled file takes 35x longer.

The JLS file format handles real-time streaming signals, and also stores a reduced waveform with statistics.  This reduced information allows that 2.71 display of a 23 GB file.  When you downsample in the Joulescope UI, the JLS file contains calibrated floats rather than raw samples.  We need to investigate why we are not getting the same speedup.

gideon-decr, 2020-10-20:
I see. So your slowdown nominally comes from switching from ints to floats. 

Consider using fixed point instead of floating point. Do you really need floating point? If so why? Floating point is heavy and comes with a lot of baggage most people are unaware of.
If you use 64 bit fixed point( or 128 bit worst-case) fixed point I am guessing you will see an unbelievable speedup. 
You can use fixed point the same way you use floating point (or any int really) except for some mathematical operations you have to dynamically increase bit widths. But good libraries can do this for you in a transparent way.

This is probably a good thing to invest in regardless. You probably want to be doing fixed point in 98% of your app. Floating point is normally for operations that are likely outside the scope of what your program really needs to do. (I am assuming you are doing basic signal processing operations and pretty much nothing more.)

This might be your ticket:
https://github.com/smlgit/fpbinary
(Disclaimer: I have not used this lib.)

Actually this one looks to have to auto-width-sizing capability:
https://github.com/francof2a/fxpmath

I recommend you look back into lossless compression. You could use it not only for on-disk compression but chunked in-memory compression. It can create a speedup outside of just disk deserialization if you do things correctly. Particularly given your very stream-focused data needs. (Aka: You have no need to quickly index multiple single-width samples over large signal distances. Most operations are co-located in blocks anyway.)

mliberty1, 2020-10-20:
The existing host-side Joulescope code is mostly single-precision floating-point for current, voltage, & power.  We do use double precision for computing reductions where extra accumulator precision is required and Python Integers (infinite precision) for computing energy and charge.  The new on-instrument statistics computation is all fixed-point inside the FPGA.

gideon-decr, 2020-10-20:
Cool! If you already understand fixed point well then the transition should be pretty easy. (If you have the desire to do it.)

Yea, there is a reason that the pandas library uses timestamps in 64bit nanoseconds instead of a float like some other libraries do. The gains are enormous and as long as there aren't a lot of exotic mathematical operations going on then you just never have any overhead.

mliberty1, 2020-10-20:
FWIW, my preferred timestamp these days for embedded systems is 34Q30 (see my [embc library](https://github.com/mliberty1/embc/blob/01866fdc62008b4aadb0c7d2b6ed4dde49976bbd/include/embc/time.h#L61)) which gives nanosecond resolution and ±272 years from whatever epoch you declare.

gideon-decr, 2020-10-20:
Cool! I like that a lot! 
I might use that general fixed point time approach for my own needs soon... (like today soon)

mliberty1, 2020-11-25:
This issue was addressed in 0.9.3.  Old recordings are not affected, but all new downsampled recordings should load faster.

closed 2020-11-25 by mliberty1