jetperch/pyjoulescope_ui #105: [Feature Request] Zoom all hotkey, clear all markers hotkey

jamesl-dm, 2020-11-03:
Can we have a 'zoom out all the way' hotkey to go with the +/- hotkeys. Maybe the '*' key?
A 'remove all markers' hotkey would also be fantastic.

Reason: I get a lot of mileage out of using the GPIOs for measuring irq timing, which results in a lot of zooming in to measure a pulse width, then clearing the markers and zooming back out all the way.

Also works great for measuring radio tx / rx timing - you can time off the current ramp down of the tx pulse, all the way to when you see the rx current change as the AGC kicks in. Possibly seconds later, with high precision, which is impossible to do on many oscilloscopes. After doing this, you spend a lot of time with the scroll wheel resetting for the next measurement.
PS. Precision would be even more amazing if there was a way to set the timebase ppm in the UI.

mliberty1, 2020-11-03:
Sounds good: 
* "*" to zoom out all.
* "backspace" or "delete" to clear all markers.

Regarding precision, we do have a number of features that all want wall time, such as issue #55 to display time as UTC.  Part of the work is to synchronize timing with your PC, which can be quite accurate using NTP.  Would this work for you?

jamesl-dm, 2020-11-03:
For the timing I'm looking for microseconds on the second (to time receive window offsets / durations in LoRaWAN) - so I don't think it can be done with NTP.
It's easy enough to account for the sample clock's ppm externally though, so being able to set it in the UI would just be icing.
But if it's already parameterized as a sample rate somewhere, and all you need to do is tweak it from the commanded 2 MHz to the realized 2.000007 MHz, it would be a nice enhancement.

mliberty1, 2020-11-25:
Added in 0.9.3.

closed 2020-11-25 by mliberty1