jetperch/pyjoulescope_ui #106: How to build an exe of the ui using PyInstaller

designingSparks, 2020-11-10:
Is this at all possible?

mliberty1, 2020-11-10:
We use pyinstaller to create the Joulescope UI executable today, and Inno Setup to create the installer.  The [joulescope.spec](https://github.com/jetperch/pyjoulescope_ui/blob/master/joulescope.spec) file contains the pyinstaller directives.  To build the executable and installer, clone this repo, cd into the directory, and type:

    pip3 install -U -r requirements.txt
    python setup.py sdist
    pyinstaller joulescope.spec

If you mean a single-file pyinstaller executable, perhaps.  The pyinstaller single file executables are a little odd as they really unzip into a temporary directory which just makes startup time longer.  

Does this answer your question?

designingSparks, 2020-11-11:
I'm just trying this. When I run the following line:
python setup.py sdist` 
the following error is thrown:
File "setup.py", line 44, in qt_rcc_path
    raise ValueError('Could not find rcc executable')
ValueError: Could not find rcc executable

mliberty1, 2020-11-11:
Due to recent issues and changes in both Qt and PySide2, the Joulescope UI is very picky about the python version and PySide2 version.  You appear to either not have PySide2 installed or an incompatible version.  You need Python 3.8.6, 64-bit (32-bit not supported) and PySide2 5.15.1.  We have not yet tested with Python 3.9.x.

If I run qt_rcc_path() in an interpreter like this:

    import os
    def qt_rcc_path():
        # As of PySide2 5.15.0, the pyside2-rcc executable ignores the --binary flag
        import PySide2
        path = os.path.dirname(PySide2.__file__)
        fname = [n for n in os.listdir(path) if n.startswith('rcc')]
        if len(fname) != 1:
            raise ValueError('Could not find rcc executable')
        return os.path.join(path, fname[0])
    qt_rcc_path()

I see:
    'C:\\bin\\Python3_8_6\\lib\\site-packages\\PySide2\\rcc.exe'

mliberty1, 2020-11-16:
Hi @designingSparks Were you able to successfully build using python 3.8.6?  Can I close this issue?

designingSparks, 2020-11-17:
Hi @mliberty1 I upgraded to Python 3.8.6 and was able to generate the standalone application using pyinstaller. However, I did encounter the following strange error message. Nevertheless, the executable appears to work.
42692 INFO: Building COLLECT because COLLECT-00.toc is non existent
42694 INFO: Building COLLECT COLLECT-00.toc
52387 INFO: Building COLLECT COLLECT-00.toc completed successfully.
Traceback (most recent call last):
  File "c:\python\python3.8.6\lib\runpy.py", line 194, in _run_module_as_main
  File "c:\python\python3.8.6\lib\runpy.py", line 87, in _run_code
  File "C:\Python\Python3.8.6\Scripts\pyinstaller.exe\__main__.py", line 7, in <module>
  File "c:\python\python3.8.6\lib\site-packages\PyInstaller\__main__.py", line 114, in run
  File "c:\python\python3.8.6\lib\site-packages\PyInstaller\__main__.py", line 65, in run_build
  File "c:\python\python3.8.6\lib\site-packages\PyInstaller\building\build_main.py", line 720, in main
  File "c:\python\python3.8.6\lib\site-packages\PyInstaller\building\build_main.py", line 667, in build
  File "joulescope.spec", line 163, in <module>
  File "c:\python\python3.8.6\lib\subprocess.py", line 489, in run
  File "c:\python\python3.8.6\lib\subprocess.py", line 854, in __init__
  File "c:\python\python3.8.6\lib\subprocess.py", line 1307, in _execute_child
FileNotFoundError: [WinError 2] The system cannot find the file specified

mliberty1, 2020-11-17:
That error is because you do not have [Inno Setup 6](https://jrsoftware.org/isinfo.php) installed.  Even if you did, you may also have to comment out the SignTool line in joulescope.iss, unless you configure code signing for your computer.

I am going to close this issue, but feel free to reopen it if you would like assistance.

closed 2020-11-17 by mliberty1