jetperch/pyjoulescope_ui #107: Problem installing joulescope on Windows 10

designingSparks, 2020-11-11:
I get a strange error message: 'An error occurred when trying to read the source file: access is denied.'  At this point the installer is trying to extract joulescope.exe to  C:\Program Files (x86)\Joulescope. I have tried running the installer as an administrator but this doesn't help.

mliberty1, 2020-11-11:
Hi @designingSparks : Are you trying to install the official Joulescope UI distribution available from [joulescope.com](https://www.joulescope.com/pages/downloads) or one that you created, as in #106.

designingSparks, 2020-11-11:
Hi @mliberty1  I'm trying to install the official Joulescope UI distribution. I still have to try creating my own as described in #106.

mliberty1, 2020-11-11:
Ok.  We have to figure out what is different about your computer.  The Joulescope UI has been installed on many, many Windows 10 machines without any issues.  Here are some things to try:

1. It's possible that the exe file was somehow corrupted.  Delete the existing installer and [download](https://www.joulescope.com/pages/downloads) a new copy.  Try to install again.
2. Are you running on a work computer administered by your IT department?  Do they have any permissions settings that restrict program installation?
3. Is the installer running from your local drive (HDD or SSD)? 
4. Are you running Win 10x64?  Note that the Joulescope UI will not run on 32-bit Windows.

designingSparks, 2020-11-11:
Hi @mliberty1 I have tried downloading the installer again but this doesn't appear to help. I have even tried a different install location. I am running on a managed work computer but I generally don't have problems installing programs since I have admin rights. I am running the installer from my local SSD drive.

designingSparks, 2020-11-11:
It is possible that the antivirus SW running in the background is blocking this.

mliberty1, 2020-11-16:
Hi @designingSparks - Any updates?  Did turning off antivirus work?

designingSparks, 2020-11-17:
I wasn't able to turn off the antivirus software but I can now build a standalone executable. It may be worth placing a standalone version of the software on the website for this reason.

closed 2020-11-17 by designingSparks