jetperch/pyjoulescope_ui #110: Capture length won't go below 1M points

jamesl-dm, 2020-11-26:
If you choose a low sampling rate like 10 Hz, and ask for a 1 hour buffer, you'll instead get a 1M point = 100k second = 27 hour buffer. Likewise for all the other settings - the buffer is always at least 1M points. Which is perfectly ok if there are technical reasons, since its easy enough to just pick a higher sampling rate. Just pointing it out in case this isn't the intended behaviour, since it had me confused for a bit.

mliberty1, 2020-11-26:
Yes, due to how pyjoulescope and pyjoulescope_ui work, they currently always allocate a buffer with at least 1 million samples.  Therefore, the minimum duration may be longer than what you request, especially for the lower sampling rates.  We certainly could adjust this behavior.

The larger issue is that it is not easy to convert between duration and RAM.  I think that the right solution is to have the user specify the RAM buffer size, and then the UI figures out the duration for the given sampling rate.  Would this make more sense to you?

jamesl-dm, 2020-11-27:
Yes, having the RAM buffer size visible seems like a good idea, so people don't choose absurdly long capture lengths and then wonder why their PC has crashed.

>and then the UI figures out the duration for the given sampling rate
I assume you mean doing this in the settings dialog itself, so you can see it update as you flip through different buffer sizes. I think that would be great.

mliberty1, 2023-03-10:
We introduced a new memory buffer management method with 1.0.0.  You now allocate buffer by RAM rather than by duration.

[image]

With the default settings, you will still get a huge buffer at 10 Hz by default.  The existing widget is designed for easily selecting sizeable buffers.  It does not have a way to manually entire small buffers.  We could add that feature if it ends up being useful.

closed 2023-03-10 by mliberty1

jamesl-dm, 2023-03-13:
UI is looking sharp - nice!