jetperch/pyjoulescope_ui #111: Clear Accumulator Blocks Streaming

andrewgoz, 2020-12-02:
In version 0.9.3 of the app (Windows), when I select Clear Accumulator it changes from "Energy XXX J in XXX s" to just "0 s" and stops updating. If already streaming that continues, but if it is stopped or is already stopped, it cannot be restarted without closing and restarting the app.

mliberty1, 2020-12-02:
Thank you for taking the time to report this issue.  I can duplicate this behavior by clicking on **Tools** → **Clear Accumulator**.  The log message shows:

```
INFO:2020-12-02 12:54:26,696:main.py:749:joulescope_ui.main:_accumulators_reset
ERROR:2020-12-02 12:54:26,727:main.py:592:joulescope_ui.main:resync queue failed
Traceback (most recent call last):
  File "joulescope_ui\main.py", line 588, in event
  File "joulescope_ui\main.py", line 655, in _on_device_statistic
TypeError: 'bool' object is not subscriptable
ERROR:2020-12-02 12:54:27,226:main.py:592:joulescope_ui.main:resync queue failed
Traceback (most recent call last):
  File "joulescope_ui\main.py", line 588, in event
  File "joulescope_ui\main.py", line 655, in _on_device_statistic
TypeError: 'bool' object is not subscriptable
```

We will fix this issue in the next release.

As a workaround for now, you can right-click on the Energy field in the Control Widget, then select **Clear**.

mliberty1, 2021-04-14:
Fixed in [v0.9.7](https://github.com/jetperch/pyjoulescope_ui/releases/tag/v0.9.7).

closed 2021-04-14 by mliberty1