jetperch/pyjoulescope_ui #112: Joulescope 0.9.3 wont work on MacOs 10.15.7

llllVIIII, 2020-12-04:
Hi,

when trying to start joulscope it wont start with an older MacOs version. Heres the error I get.

>  $> /Applications/joulescope.app/Contents/MacOS/joulescope_launcher
> [5877] Error loading Python lib '/Applications/joulescope.app/Contents/MacOS/Python': dlopen: dlopen(/Applications/joulescope.app/Contents/MacOS/Python, 10): Symbol not found: _preadv
>   Referenced from: /Applications/joulescope.app/Contents/MacOS/Python (which was built for Mac OS X 11.0)
>   Expected in: /usr/lib/libSystem.B.dylib
>  in /Applications/joulescope.app/Contents/MacOS/Python

mliberty1, 2020-12-04:
Yes, Apple has made life very difficult for developers.  From our perspective, one Mac platform has now become three:

1. macOS 11.0.1 (Big Sur) on AMD64
2. macOS 10.x on AMD64
3. macOS 11.0.1 (Big Sur) on ARM

The official 0.9.3 release only supports (1).  Since you are running on (2), you have a few choices:

1. Run Joulescope UI 0.9.2 (which does not run on macOS 11, see #108 and the [forum](https://forum.joulescope.com/t/joulescope-ui-0-9-2-and-macos-big-sur/317)).
2. Wait until we figure out how to support macOS 10.  We may end up dropping official release support in favor of (3).
3. Run from homebrew.  

If you choose to run from homebrew, you will be using the Joulescope python packages available on pypi.  First, install [brew](https://brew.sh/) on your machine. Open a terminal, and type:

    brew update
    brew upgrade
    pip3 install -U  joulescope_ui PySide2

You should then be able to run the Joulescope UI using:

    python3 -m joulescope

Do any of these options work for you?

llllVIIII, 2020-12-04:
Thanks for the quick replay! 

For now I have reverted to Joulescope 0.9.2. I didn't know that homebrew was also an option. Maybe I will give it a shot later.

mliberty1, 2020-12-04:
Reopening.  Although we have workarounds, we still need to determine the official course of action.

mliberty1, 2021-04-14:
After investigating, we have no easy answer.  We will distribute two macOS binaries for the upcoming release.  The two versions will be:

* macOS 10 x86_64
* macOS 11 x86_64

macOS 11 arm64 will run under Rosetta using the x86_64 binaries.  We will punt on creating [universal binaries](https://developer.apple.com/documentation/apple-silicon/building-a-universal-macos-binary) through pyinstaller.

mliberty1, 2021-04-14:
Address in [v0.9.7](https://github.com/jetperch/pyjoulescope_ui/releases/tag/v0.9.7) with two download options.

closed 2021-04-14 by mliberty1