jetperch/pyjoulescope_ui #114: 0.9.3 exception

krbvroc1, 2021-01-16:
Python 3.8.5 under Ubuntu 20.04 LTS

When I click Tools->Clear accumulator the program no longer works.

```
ERROR:2021-01-16 14:57:26,392:command_processor.py:384:joulescope_ui.command_processor:subscriber error for topic=Device/#state/play, value=True
Traceback (most recent call last):
  File "/home/x/python-virtual-environments/joulescope/lib/python3.8/site-packages/joulescope_ui/command_processor.py", line 382, in _subscriber_call
    fn(topic, value)
  File "/home/x/python-virtual-environments/joulescope/lib/python3.8/site-packages/joulescope_ui/main.py", line 1248, in _on_device_state_play
    self._device_stream_start()
  File "/home/x/python-virtual-environments/joulescope/lib/python3.8/site-packages/joulescope_ui/main.py", line 1211, in _device_stream_start
    self._accumulators_zero_last()
  File "/home/x/python-virtual-environments/joulescope/lib/python3.8/site-packages/joulescope_ui/main.py", line 764, in _accumulators_zero_last
    for z in self._accumulators['fields'].values():
TypeError: 'bool' object is not subscriptable

```

mliberty1, 2021-01-17:
Hi @krbvroc1, and I apologize that you ran into this issue.  As a workaround for now, you can right-click on the Energy field in the Control Widget, then select **Clear**.

We are tracking it as #111, and it has already been fixed in the "develop" branch.  The fix will be in our next release.

I am going to close this issue as a duplicate, but please feel don't hesitate to comment or reopen it if you have questions.  Thanks again for taking the time to report this issue!

closed 2021-01-17 by mliberty1

krbvroc1, 2021-01-17:
@mliberty1 - Thanks for the response. You mentioned a workaround of right clicking the Energy field and selecting Clear. Overnight I ran a test for 12 hours and this morning simply clicked the Energy field in the top right corner. I was expecting it to maybe toggle the units or something non-destructive, but instead it reset the value and I lost my in progress test. Is this expected behavior?

mliberty1, 2021-01-17:
Yes, left-clicking on it will cause it to reset.  Right-clicking will give you options.  If you hit this accidentally, you can undo this *Clear* operation (and many other operations) using your platform-specific undo key combination.  On Windows, you use Ctrl-Z.

krbvroc1, 2021-01-17:
@mliberty1 Good to know. My suggestion would be to add a Clear to the right-click options so a left-click is not destructive. Thanks.