jetperch/pyjoulescope_ui #115: joulescope fails on startup with Qt AttributeError on macOS 10.15

jeremyherbert, 2021-01-21:
I have installed joulescope using on macOS 10.15 with these commands:

```
pip install -U joulescope joulescope_ui PySide2
pip install --force-reinstall PySide2==5.13.2
```

However this does not work when running `python -m joulescope`

```
*****************************
* Joulescope User Interface *
* UI Version = 0.9.3        *
* Driver Version = 0.9.3    *
*****************************

info = {
  "joulescope": {
    "ui_version": "0.9.3",
    "driver_version": "0.9.3"
  },
  "platform": {
    "name": "darwin",
    "python_version": "3.7.4 (default, Aug 13 2019, 15:17:50) \n[Clang 4.0.1 (tags/RELEASE_401/final)]",
    "platform": "Darwin-19.6.0-x86_64-i386-64bit",
    "processor": "i386",
    "executable": "/Users/jeremy/opt/anaconda3/bin/python",
    "frozen": false,
    "paths": {
      "dirs": {
        "app_path": "/Users/jeremy/Library/Application Support/joulescope",
        "config": "/Users/jeremy/Library/Application Support/joulescope/config",
        "log": "/Users/jeremy/Library/Application Support/joulescope/log",
        "firmware": "/Users/jeremy/Library/Application Support/joulescope/firmware",
        "themes": "/Users/jeremy/Library/Application Support/joulescope/themes",
        "data": "/Users/jeremy/Documents/joulescope"
      },
      "files": {
        "config": "/Users/jeremy/Library/Application Support/joulescope/config/joulescope_config.json"
      }
    }
  }
}

=====
Traceback (most recent call last):
  File "/Users/jeremy/opt/anaconda3/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/Users/jeremy/opt/anaconda3/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/Users/jeremy/opt/anaconda3/lib/python3.7/site-packages/joulescope/__main__.py", line 22, in <module>
    sys.exit(run())
  File "/Users/jeremy/opt/anaconda3/lib/python3.7/site-packages/joulescope/entry_points/runner.py", line 97, in run
    return ui_entry_point.run()
  File "/Users/jeremy/opt/anaconda3/lib/python3.7/site-packages/joulescope_ui/main.py", line 1916, in run
    ui.run(filename)
  File "/Users/jeremy/opt/anaconda3/lib/python3.7/site-packages/joulescope_ui/main.py", line 567, in run
    self._on_widgets_active('Widgets/active', self._cmdp['Widgets/active'])
  File "/Users/jeremy/opt/anaconda3/lib/python3.7/site-packages/joulescope_ui/main.py", line 1516, in _on_widgets_active
    self._widgets_add(widget)
  File "/Users/jeremy/opt/anaconda3/lib/python3.7/site-packages/joulescope_ui/main.py", line 1454, in _widgets_add
    dock_widget = MyDockWidget(self, widget_def, self._cmdp, instance_id)
  File "/Users/jeremy/opt/anaconda3/lib/python3.7/site-packages/joulescope_ui/main.py", line 217, in __init__
    self.inner_widget = widget_def['class'](self, cmdp, self.state_preference)
  File "/Users/jeremy/opt/anaconda3/lib/python3.7/site-packages/joulescope_ui/widgets/control/control_widget.py", line 200, in __init__
    self._accumLabel.setTextFormat(QtGui.Qt.RichText)
AttributeError: module 'PySide2.QtGui' has no attribute 'Qt'
```

jeremyherbert, 2021-01-21:
I am definitely running the right version according to ipython:

```
Python 3.7.4 (default, Aug 13 2019, 15:17:50)
Type 'copyright', 'credits' or 'license' for more information
IPython 7.17.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import PySide2

In [2]: PySide2.__version__
Out[2]: '5.13.2'

In [3]: import PySide2.QtGui

In [4]: PySide2.QtGui.Qt
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-4-ffab4a7797dc> in <module>
----> 1 PySide2.QtGui.Qt

AttributeError: module 'PySide2.QtGui' has no attribute 'Qt'

In [5]: dir(PySide2.QtGui)
Out[5]:
['QAbstractOpenGLFunctions',
 'QAbstractTextDocumentLayout',
 'QAccessible',
 'QAccessibleEditableTextInterface',
 'QAccessibleEvent',
 'QAccessibleInterface',
 'QAccessibleObject',
 'QAccessibleStateChangeEvent',
 'QAccessibleTableCellInterface',
 'QAccessibleTableModelChangeEvent',
 'QAccessibleTextCursorEvent',
 'QAccessibleTextInsertEvent',
 'QAccessibleTextInterface',
 'QAccessibleTextRemoveEvent',
 'QAccessibleTextSelectionEvent',
 'QAccessibleTextUpdateEvent',
 'QAccessibleValueChangeEvent',
 'QAccessibleValueInterface',
 'QActionEvent',
 'QBackingStore',
 'QBitmap',
 'QBrush',
 'QClipboard',
 'QCloseEvent',
 'QColor',
 'QConicalGradient',
 'QContextMenuEvent',
 'QCursor',
 'QDesktopServices',
 'QDoubleValidator',
 'QDrag',
 'QDragEnterEvent',
 'QDragLeaveEvent',
 'QDragMoveEvent',
 'QDropEvent',
 'QEnterEvent',
 'QExposeEvent',
 'QFileOpenEvent',
 'QFocusEvent',
 'QFont',
 'QFontDatabase',
 'QFontInfo',
 'QFontMetrics',
 'QFontMetricsF',
 'QGradient',
 'QGuiApplication',
 'QHelpEvent',
 'QHideEvent',
 'QHoverEvent',
 'QIcon',
 'QIconDragEvent',
 'QIconEngine',
 'QImage',
 'QImageIOHandler',
 'QImageReader',
 'QImageWriter',
 'QInputEvent',
 'QInputMethod',
 'QInputMethodEvent',
 'QInputMethodQueryEvent',
 'QIntValidator',
 'QKeyEvent',
 'QKeySequence',
 'QLinearGradient',
 'QMatrix',
 'QMatrix2x2',
 'QMatrix2x3',
 'QMatrix2x4',
 'QMatrix3x2',
 'QMatrix3x3',
 'QMatrix3x4',
 'QMatrix4x2',
 'QMatrix4x3',
 'QMatrix4x4',
 'QMouseEvent',
 'QMoveEvent',
 'QMovie',
 'QNativeGestureEvent',
 'QOffscreenSurface',
 'QOpenGLBuffer',
 'QOpenGLContext',
 'QOpenGLContextGroup',
 'QOpenGLDebugLogger',
 'QOpenGLDebugMessage',
 'QOpenGLExtraFunctions',
 'QOpenGLFramebufferObject',
 'QOpenGLFramebufferObjectFormat',
 'QOpenGLFunctions',
 'QOpenGLPixelTransferOptions',
 'QOpenGLShader',
 'QOpenGLShaderProgram',
 'QOpenGLTexture',
 'QOpenGLTextureBlitter',
 'QOpenGLTimeMonitor',
 'QOpenGLTimerQuery',
 'QOpenGLVersionProfile',
 'QOpenGLVertexArrayObject',
 'QOpenGLWindow',
 'QPageLayout',
 'QPageSize',
 'QPagedPaintDevice',
 'QPaintDevice',
 'QPaintDeviceWindow',
 'QPaintEngine',
 'QPaintEngineState',
 'QPaintEvent',
 'QPainter',
 'QPainterPath',
 'QPainterPathStroker',
 'QPalette',
 'QPdfWriter',
 'QPen',
 'QPicture',
 'QPictureIO',
 'QPixelFormat',
 'QPixmap',
 'QPixmapCache',
 'QPointingDeviceUniqueId',
 'QPolygon',
 'QPolygonF',
 'QPyTextObject',
 'QQuaternion',
 'QRadialGradient',
 'QRasterWindow',
 'QRawFont',
 'QRegExpValidator',
 'QRegion',
 'QResizeEvent',
 'QScreen',
 'QScrollEvent',
 'QScrollPrepareEvent',
 'QSessionManager',
 'QShortcutEvent',
 'QShowEvent',
 'QStandardItem',
 'QStandardItemModel',
 'QStaticText',
 'QStatusTipEvent',
 'QStyleHints',
 'QSurface',
 'QSurfaceFormat',
 'QSyntaxHighlighter',
 'QTabletEvent',
 'QTextBlock',
 'QTextBlockFormat',
 'QTextBlockGroup',
 'QTextBlockUserData',
 'QTextCharFormat',
 'QTextCursor',
 'QTextDocument',
 'QTextDocumentFragment',
 'QTextDocumentWriter',
 'QTextFormat',
 'QTextFragment',
 'QTextFrame',
 'QTextFrameFormat',
 'QTextImageFormat',
 'QTextInlineObject',
 'QTextItem',
 'QTextLayout',
 'QTextLength',
 'QTextLine',
 'QTextList',
 'QTextListFormat',
 'QTextObject',
 'QTextObjectInterface',
 'QTextOption',
 'QTextTable',
 'QTextTableCell',
 'QTextTableCellFormat',
 'QTextTableFormat',
 'QToolBarChangeEvent',
 'QTouchDevice',
 'QTouchEvent',
 'QTransform',
 'QValidator',
 'QVector2D',
 'QVector3D',
 'QVector4D',
 'QWhatsThisClickedEvent',
 'QWheelEvent',
 'QWindow',
 'QWindowStateChangeEvent',
 '__doc__',
 '__file__',
 '__loader__',
 '__name__',
 '__package__',
 '__spec__',
 'qAlpha',
 'qApp',
 'qBlue',
 'qGray',
 'qGreen',
 'qIsGray',
 'qRed',
 'qRgb',
 'qRgba']
```

jeremyherbert, 2021-01-21:
It seems that this import is wrong, it should be coming from `QtCore` rather than `QtGui`. But surely I am not the only one with this problem then?

```
Python 3.7.4 (default, Aug 13 2019, 15:17:50)
Type 'copyright', 'credits' or 'license' for more information
IPython 7.17.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import PySide2.QtCore

In [2]: PySide2.QtCore.Qt
Out[2]: PySide2.QtCore.Qt

In [3]: PySide2.QtCore.Qt.RichText
Out[3]: PySide2.QtCore.Qt.TextFormat.RichText
```

jeremyherbert, 2021-01-21:
Replacing the line: https://github.com/jetperch/pyjoulescope_ui/blob/ee814d50683029238013cc48d9144d8f06449eaa/joulescope_ui/widgets/control/control_widget.py#L200

with 

```python
self._accumLabel.setTextFormat(QtCore.Qt.RichText)
```

fixes the problem and the UI can then start up. But I'm still confused as to how this is possible?

mliberty1, 2021-01-21:
Hi @jeremyherbert.  Thank you very much for investigating and the detailed information!  I took a look at the Qt documentation, and [QtGui.Qt](https://doc.qt.io/qtforpython-5/PySide2/QtGui/Qt.html) inherits from [QtCore.Qt](https://doc.qt.io/qtforpython-5/PySide2/QtCore/Qt.html).  You are using PySide2 5.13.2, released on [Nov 1, 2019](https://pypi.org/project/PySide2/#history), which we pinned for python earlier than 3.8.  We have been testing on Python 3.8 and 3.9 with PySide2 5.15.  Using vanilla Python 3.7.9 on Windows:

[image]

which shows that it is actually calling out PySide2.QtCore.Qt.TextFormat.RichText.  I have no idea why your install does not import PySide2.QtGui.Qt.  Anaconda is definitely quirky, so that may explain the difference.  pyqtgraph also does [some strange things](https://github.com/pyqtgraph/pyqtgraph/blob/master/pyqtgraph/Qt.py) to the Qt imports so that it works across all different version.

Regardless, we can directly use QtCore.Qt.  I grep'ed the codebase and found:
[image]

I'll push up a change to the "develop" branch momentarily.

mliberty1, 2021-01-21:
Change pushed!  You should be able to run without issues from the **develop** branch.  I am going to close the issue, but feel free to comment below.  Also, reopen it if you run into problems.  Thanks!

closed 2021-01-21 by mliberty1