jetperch/pyjoulescope_ui #122: Feature request: when adding dual cursors, display mAh as well

limpkin, 2021-04-10:
... because dividing by 3.6 is too hard ;)

mliberty1, 2021-04-10:
Hi @limpkin - I think that we already have this!  Check this out:

[image]

You need to change the units in preferences.  Here's how:

1. Select **File** → **Preferences**
2. Set **Profile** to defaults.
3. Select **Units**.  
4. Set **charge** to **Ah**.

[image]

Then, the charge display on the multimeter widget, the single value widget, the control widget, the right-hand statistics on the waveform view and the dual markers should all be in Ah, not C.

Does this do what you want?

mliberty1, 2021-04-16:
Hi @limpkin - I am following up to see if I can close this issue.  Does my response above address what you want?

limpkin, 2021-04-16:
sorry, notifications are not working for me with github :/.
yeah of course, please close it :) thanks!!!

closed 2021-04-16 by limpkin