jetperch/pyjoulescope_ui #128: C binary compatibility error after pip upgrade on mac

jeremyherbert, 2021-05-04:
Hi,

I'm running mac os 10.15.6, and had joulescope working fine previously on 0.9.3. However, after running `pip install --upgrade joulescope-ui` I get the following output when trying to run joulescope:

```
$ python -m joulescope
Traceback (most recent call last):
  File "/Users/jeremy/opt/anaconda3/lib/python3.7/runpy.py", line 183, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/Users/jeremy/opt/anaconda3/lib/python3.7/runpy.py", line 142, in _get_module_details
    return _get_module_details(pkg_main_name, error)
  File "/Users/jeremy/opt/anaconda3/lib/python3.7/runpy.py", line 109, in _get_module_details
    __import__(pkg_name)
  File "/Users/jeremy/opt/anaconda3/lib/python3.7/site-packages/joulescope/__init__.py", line 15, in <module>
    from joulescope.driver import scan, scan_require_one, scan_for_changes, \
  File "/Users/jeremy/opt/anaconda3/lib/python3.7/site-packages/joulescope/driver.py", line 20, in <module>
    from joulescope.stream_buffer import StreamBuffer, DownsamplingStreamBuffer
  File "joulescope\stream_buffer.pyx", line 1, in init joulescope.stream_buffer
ValueError: numpy.ndarray size changed, may indicate binary incompatibility. Expected 88 from C header, got 80 from PyObject
```

Is it possible that the numpy version required by the package metadata is too low? I have 1.19.1 which is higher than the required one:

```
...
Requirement already satisfied, skipping upgrade: numpy>=1.16 in ./opt/anaconda3/lib/python3.7/site-packages (from pyjls>=0.3.1->joulescope-ui) (1.19.1)
...
```

jeremyherbert, 2021-05-04:
I just completely reinstalled python to 3.9.5 (with numpy 1.20.2) and it is working now. Feel free to close this but it would still be good to know where this came from.

mliberty1, 2021-05-04:
Well, the numpy C API changed in numpy 1.20.  Check out this [StackOverflow](https://stackoverflow.com/questions/66060487/valueerror-numpy-ndarray-size-changed-may-indicate-binary-incompatibility-exp) thread.  The pyjoulescope library uses Cython to compile C & Cython code that depends on the numpy header files.  Is it possible that the compilation found a different set of numpy header files on your machine during the build process?  Anaconda is another whole issue that adds complexity.

If I were to start troubleshooting this issue, I would uninstall joulescope and force reinstall cleanly, like this:
```
pip3 uninstall -y joulescope pyjls joulescope_ui
pip3 install -U numpy
pip3 install -U --no-cache-dir joulescope_ui
```

An alternative approach is to create a new virtual environment.

I think that pyjoulescope and the pyjoulescope_ui should still run on Python 3.7, but we (along with numpy) are [dropping support soon](https://numpy.org/neps/nep-0029-deprecation_policy.html) - Dec 26, 2021.

Upgrading to python 3.9 is definitely the right way to go!

closed 2021-05-04 by mliberty1