jetperch/pyjoulescope_ui #130: Configure display precision for waveform markers

mliberty1, 2021-06-18:
Version: 0.9.7
Platform: All

**Description**
The waveform view always displays 6 digits for all numbers, like this:
[image]

The summary over the waveform shown at right also displays 6 digits.  

This takes up more room than needed if you only care to see 3 digits.  The extra precision can also detract from presentations and feel overwhelming.

**Expected behavior**
Add a configuration option that would allow the user to set the precision.  You can still leave 6 digits as the default, but then a user could configure 3 digits if that is all they want.

atsju, 2022-04-27:
In the same way for "optionalization"an maybe linked to #144 it could be great to choose which info to display.
For example in your pricture I'm only interested into first and two last lines. The rest is only distraction for the eyes.

mliberty1, 2024-02-28:
Added in 1.0.58.  Right-click on the waveform, **Settings** → **precision**.

closed 2024-02-28 by mliberty1