jetperch/pyjoulescope_ui #137: add audio annotation feature

raveslave, 2021-09-27:
far fetched.. but incredibly useful!   

add a feature to record a mono audio track (can be compressed) while doing measurements   
sometimes you have both hands buzy, pushing buttons, turning knobs, then it would be a great help to add voice-snippets to the time-line so you can listen back on what was done while capturing.

mliberty1, 2021-09-27:
This feature is not so far-fetched.  With the original JLS v1 format, recording audio alongside Joulescope data would be very difficult.  However, the new [JLS v2](https://github.com/jetperch/jls) file format should easily be able to store a microphone input audio track.  The Joulescope UI already supports loading & displaying JLS v2 files, but it does not yet support recording them.

The existing JLS v2 implementation only supports 32-bit float data values, but we plan on adding other formats.  The "Signed integers in nibble (4 bit) increments" set to 16-bit would be good for audio.  JLS v2 tracks also contain wall-clock time synchronization which would keep the Joulescope data synchronized to the audio.  

New recording features:
1. Configure recording.  Optionally enable microphone audio input.
2. Select & open audio input device
3. Save audio to JLS v2 track

On playback:
1. Display audio waveform.  Min/max already would display the sound envelope so that you can easily find audio events.
2. Ability to play back the audio waveform from a starting point.  Could use the start of the visible window, marker, and/or current cursor location.

The JLS v2 format also supports annotations.  While these can be saved in the original JLS file, you normally want them in a separate JLS file so that you can edit the annotations without affecting the original data.  We could potentially perform speech recognition and automatically generate text annotations.

raveslave, 2021-09-27:
great summary (re 1,2,3), the audio track could benefit from having a lower configurable threshold so that annotations could represent the start of new voice-overs (with a reasonable 'release' of ~2-3 sec silence to avoid multiple 'regions' being created)