jetperch/pyjoulescope_ui #145: Add energy to dual marker stats

blackenstock, 2022-03-21:
Dual marker stat is missing the energy value for the given time range.

mliberty1, 2022-03-21:
Hi @blackenstock!

The dual markers display the integration for current and power, which is charge and energy, respectively.  Here is an example:
[image]

In order to display energy, you do need to enable the `power` waveform.  Does this work for you?

The information displayed with the dual markers is not very customizable at this time with Joulescope UI 0.9.11.  However, we do plan to make some improvements this year.  Check out [this forum post](https://forum.joulescope.com/t/cursor-times-and-delta-t-enhancement/427/2) which contains a summary of all these issues.

mliberty1, 2022-03-24:
Hi @blackenstock - Did my previous post address the issue?  Since I haven't heard a reply, I am going to assume so and close this issue.  Please reopen it if the response does not fully address the issue!

closed 2022-03-24 by mliberty1