jetperch/pyjoulescope_ui #147: Minor display issue

atsju, 2022-04-26:
Version 0.9.11

Not very important but not perfect

[image]

mliberty1, 2022-04-26:
Hi @atsju and thanks for submitting this issue.  This seems to be a Qt issue with some platforms.  My Win 11 amd64 platform renders correctly.  We will be moving from PySide2/Qt4 to PySide6/Qt6 later this year, so I am sure we will have lots of issues like this to investigate.  Let's keep this issue open as a reminder to double-check all of our "special" style sheet items.

mliberty1, 2023-03-10:
I guess that this is "fixed" in 1.0.0 since we removed Preferences.  When we add back the new settings widget, we will make sure that it looks right.  If you see this issue again when we add settings, please reopen this issue!

closed 2023-03-10 by mliberty1