jetperch/pyjoulescope_ui #148: Issue template

atsju, 2022-04-26:
Hi,
It seems like the main entry point for your product is the forum and then mostly you @mliberty1 create issues. 
If it's also allowed to add issues/questions directly to the Github by users then maybe you should consider adding an [issue template](https://docs.github.com/en/communities/using-templates-to-encourage-useful-issues-and-pull-requests/configuring-issue-templates-for-your-repository). The minimum to ask about would be the UI version.

I'm embedded FW developper and used to open source. If you have any question or remark, feel free to ask.

mliberty1, 2022-04-26:
Thanks for the feedback @atsju .  It's perfectly fine submitting issues directly to our open source projects here on GitHub.  Many Joulescope customers are more comfortable posting to the forum.  As you noticed, I then often create a corresponding issue in the right project.  

I have not spent time configuring our projects on GitHub.  The "Issue Forms" beta looks interesting.  I would want to collect the Joulescope UI version and platform.  As you can see from my issues, I tend to have at least two sections: **Description** and **Expected Behavior**.

Our bug volume and number of collaborators have both been fairly low, so the existing process has been good enough.  However, I should set aside some time to improve things!  I'll keep this open as a reminder.

atsju, 2022-04-26:
Thanks a lot for all your feedbacks on all issues.  
Consider adding "good first issue" to this one and #146. It's used to flag easy improvements for external contributors. 
I didn't know the "issue template", it seems pretty cool. I might create the thing if I find time and motivation.

mliberty1, 2022-04-27:
Thank you @atsju for this improvement!

PR #150 (2022-04-27): add issue form
closes #148 

You can check how it renders here https://github.com/atsju/wikiTest/issues
I plan to add more like "question" and "enhancement"
Also it seems the label `triage` must exist or it will not be added
instead of asking for SW version we could also have a checkbox requesting to having tested with latest XXX version (XXX being in the issue text at the end)

Also, default branch is `master` this leads to PR being done to master by default which seems wrong because you have a develop branch. Maybe you want to remove the develop branch. Maybe you prefer to switch the default [... cut]
closed 2022-04-27 by mliberty1