jetperch/pyjoulescope_ui #149: Github actions for maintenance

atsju, 2022-04-26:
This tool has a huge quality level. v0.9.11 never crashed once in several weeks of use. To maintain this level of quality and for ease of maintenance, I would recommend having some sort of CI here.   
I see you already have CI on https://github.com/jetperch/pyjoulescope using Travis so I will not explain what CI is.

mliberty1, 2022-04-26:
Thank @atsju !  Yes, we do a **LOT** of manual Joulescope UI testing on each release to keep our quality level high.  Many of these tests are less work to do manually than to automate with CI, at least for now.  

However, I should definitely set up the unit tests to run on GitHub Actions.  Our newer [JLS v2](https://github.com/jetperch/jls/actions) project uses GitHub actions across multiple platforms, so we now are familiar with this.  I have considered setting up a worker farm of Win 11, macOS, and Linux and using [self-hosted runners](https://docs.github.com/en/actions/hosting-your-own-runners/about-self-hosted-runners), each with attached Joulescopes.  We already have this setup for our manual release testing, and it would be great to perform at least a key subset of our manual tests using CI.

We are planning some significant changes to the UI for later this year.  I think that we should get GitHub Actions running sooner rather than later.  We can then consider more improvements, such as the worker farm with real hardware, later.

mliberty1, 2023-07-25:
As of Jul 2023, using GitHub actions for CI for the pyjoulescope_ui, joulescope_driver, and jls.

closed 2023-07-25 by mliberty1