jetperch/pyjoulescope_ui #161: [Enhancement]: change µ symbol to xBar

atsju, 2022-06-08:
### UI version

0.9.11

### Your idea

Today the mean value is represented with `µ`
Please replace with `mean` or xBar because µ is not really mean value but the mean population of a distribution.
[image]

for your reference 
https://www.statisticshowto.com/probability-and-statistics/statistics-definitions/sample-mean/
https://wumbo.net/symbol/mu/#:~:text=The%20Greek%20letter%20%CE%BC%20(mu,population%20mean%20of%20a%20distribution.
https://wumbo.net/formula/population-mean/

mliberty1, 2022-06-08:
Hi @atsju - I'm not sure about this.  Disclaimer: I have done a lot of statistics in my career, including some fancy extended Kalman filtering, and things get very blurry when you capture ALL the data ;)

I think of x̄ as estimating the population mean µ of a stationary process when taking a limited number of samples from a larger set of the full population.  For example, the weight of the deer population near my house today.  I can take 3 deer, weigh them, and compute x̄ that approximates µ.  It does not approximate µ in February, since the weight of the local deer population will likely be much lower at the end of winter.  In other words, the weight of deer is close enough to a stationary process over the timespan of a week or month, but it is not a stationary process over the full year.

While your Joulescope does sample the incoming data, it does so in a mathematically rigorous way such that no data is missed within the bandwidth of interest.  Therefore, for the time window denoted by dual markers, we have ALL the data.  When we compute mean, we are exactly computing µ for that time window, at least to the precision of the instrument.  It is not an estimate based upon a subset of data.

Now, you may want to use that value to estimate the average over a larger time window or different conditions.  However, just like weighing every deer in my local population today may be a bad estimate for the deer population in the distant future, the Joulescope mean over that time window may be a good or bad estimate for other times.  Joulescope cannot presume to be measuring a stationary process either.

So, it comes down to how you are using your Joulescope.  If you view that time window as the entire event of interest, then Joulescope measures µ exactly for that time window.  This is why we display µ today.  If you view that µ as one event, such as a sensor read / transmit cycle, and you want to estimate the average for a set of events, then a single time window µ is the event estimator x̄.  Unfortunately, I believe that x̄ would be even more misleading than µ as it presumes you want to use that time window to estimate something else.

For what it's worth, I have received other feedback that I should just change these to "avg" and "std".  Since what is right depends upon what you are attempting to measure, I cannot make it strictly correct.  Someone is always going to be unhappy.

Does this make sense to you?  Given all this, would you prefer the less biased term "avg", too?

mliberty1, 2022-06-08:
For what it's worth, I just checked three pieces of equipment close at hand: a Keysight 34465A, Rigol DS1102E, and Saleae Logic Pro 16.  The Keysight displays `Average` and the other two display `Vavg`.

atsju, 2022-06-08:
You are light years above my level for statistics and I'm not sure to 100% understand what you try to say but for sure I posted too fast. With your message and some additional research it seams µ and /x could be used
https://keydifferences.com/difference-between-sample-mean-and-population-mean.html#:~:text=The%20arithmetic%20mean%20of%20the,%CE%BC%20(Greek%20term%20mu).

Now from my understanding of the article (I discovered after original question) I think you are completely right if we define the population as the measurement points between the cursors. Philosophic question : What should be the notation as the measurement samples are already an approximation of a continuous analog signal ? i.e. Shall we consider the symbol as relative the the sample points or relative to the analog signal ?

Lecroy oscilloscopes use `mean` notation. And I would also prefer to have `mean` or `avg`. From my readings, average seems to be a more accurate choice but I feel like you know this much better than I do.

Anyway, thank you very much for your detailed answer

mliberty1, 2022-06-08:
_Philosophic question : What should be the notation as the measurement samples are already an approximation of a continuous analog signal ?_

Although statistics and signal processing both have samples and the concepts are not unrelated, the outcomes are very different.  In [statistics](https://en.wikipedia.org/wiki/Sampling_(statistics)), samples are usually taken randomly from a population.  In [signal processing](https://en.wikipedia.org/wiki/Nyquist%E2%80%93Shannon_sampling_theorem), samples are traditionally taken at fixed intervals from an analog signal.  As long as the signal is bandwidth limited, the sampled analog signal can be fully reconstructed without loss (minus quantization errors and nonlinearity errors in the conversion process).  For practical purposes, we consider the digitized waveform to be equivalent to the analog waveform.  Therefore, computing the mean on the digitized signal samples is the same as computing the continuous-time mean on the analog signal.

I agree that we should change `µ` to `avg`.  We should also change `σ` to `std`, so that we have avg, std, min, max, p2p for the statistics.

mliberty1, 2023-03-10:
Changed to avg and std in 1.0.0.

[image]

closed 2023-03-10 by mliberty1