jetperch/pyjoulescope_ui #162: [Enhancement]: range in datasheet VS UI

atsju, 2022-06-15:
### UI version

0.9.11

### Your idea

During todays measurement we needed to know the guaranteed precision for used range so we displayed range in UI. But there is no direct link between range number in UI and the datasheet.

Please consider adding a column in datasheet.
[image]

mliberty1, 2022-06-15:
Thanks for the suggestion @atsju !  I can confirm that the numbering above matches what the UI displays.  Also, 7 is `off`, meaning no shunt resistor is selected.

mliberty1, 2023-03-10:
1.0.0 now applies the current range labels to the y-axis:

[image]

closed 2023-03-10 by mliberty1