jetperch/pyjoulescope_ui #165: [Enhancement]: improve waveform saving

atsju, 2022-08-25:
### UI version

0.9.11

### Your idea

let's say I want to save 1 out of 10 measurements and I need to pause at each measurement to know if I want to save.  
Today I need to click on save at **each** start of capture and then delete the 9 unneeded files.

It would be great to have 

1. A "save as" button of current buffer once you pause. Data are displayable so we have it somewhere, just add a way to save it to be able to compare/reuse it later.
2. A way to "record always" each time we start and pause, create a new file with a new name. Even better if you can set a rolling buffer with maximum number of files or maximum size on disk.

Option 1. Is a must have. Option 2 is only an idea, I do not need it personally but I suppose there could be some use.

mliberty1, 2022-08-25:
Hi @atsju 

For (1), the Joulescope UI already has several ways.  The quickest is to right-click on the center of the waveform widget, then select either **Export visible data** or **Export all data**.  
[image]

You can also add dual markers, move them to the region of interest, right-click on one of the markers, and select **Export data**.
[image]

In both cases, you can only save data still in the RAM buffer, which defaults to 30 seconds.

Does this meet your enhancement request for (1)?

---
For (2), I am not sure about this feature.  To me, it seems that it would cause more problems with only limited benefits.  Here are some concerns:

a. What happens if a user then manually starts a recording to a file?  Do we record both files?  

b. How does a customer know which automatic recording has which test case as they look back in history?  I can definitely see a use for the current capture, perhaps one or two previous.  Beyond that, automatically named files with no correlation to what the user is doing limits the usefulness of this data.

c. Some customers record long captures, and recording two files would definitely be an issue.  I can easily see someone forgetting that they enabled this feature once.  This feature presumes that customers are recording short (minutes) snippets, which is definitely not true for many of our customers.

atsju, 2022-08-25:
(1) 100% covers my need, I did not notice it before. Sorry. 
Thank you very much for support.

closed 2022-08-25 by atsju

mliberty1, 2022-08-25:
I love having feature requests that are already implemented!  However, the fact that you didn't notice this means that we need to make the feature more accessible.  We definitely need more documentation and videos for the UI.  I will think about how to make it easier to find some of these right-click features.  In the meantime, when in doubt, right-click! (control-click on mac).

atsju, 2022-08-25:
To be honest I expected it to be in "file->export" or "file->save as".  
I even discussed the point with a colleague who confirmed "no you can't save after pause like you do with recording".

An other misleading point I knew and presented to my colleague is that IN0/1 can be displayed while not being "active".  
From my point of view, when you display IN0 IN1 you must automatically map them to rear IN0/IN1. I understand they use the LSB of measurement but I don't see a case where you need them displayed but not mapped to rear.
[image]

mliberty1, 2022-08-25:
I can see why you would want it in the File menu, and for the existing UI, this makes sense when the Waveform widget is visible. 

Yeah, the actual JS110 hardware has more flexibility on GPI mapping.  We thought about exposing this flexibility through the UI, but never did.  The UI always maps the JS110 in0 to current lsb and in1 to voltage lsb.  The UI could certainly force the GPI mapping when the user selects the waveform signals.  The UI would not know to disable them though, since the user may still want them enabled for a JLS recording.  The existing recording implementation always record all signals, but this will change in the near future.

For what it's worth, the upcoming JS220 fixes this.  It only communicates the selected signals.  The GPI signals are also communicated separately, not in the current & voltage LSBs, so there is no downside to enabling them other than extra USB bandwidth.  

We will be changing how signals are selected in the UI in the near future to support the JS220.  I think that we should be able to add automatic JS110 in0 & in1 enable/disable without the separate GPIO widget.