jetperch/pyjoulescope_ui #166: [Bug]: Crash on MacOS Ventura 13.0 (22A380)

mrks115, 2022-10-24:
### UI version

0.9.11

### What OS are you seeing the problem on?

Mac OS 12 arm64 (M1 and newer)

### What happened?

App consequently crashes after updating to Ventura. Before the update everything worked fine.

### What was expected?

UI should not crash

### How to reproduce?

1. Open App on MacOS Ventura 13.0 (22A380)

mliberty1, 2022-10-24:
The macOS 13 update is not yet available for my MacBook Pro M1 yet.  The [Apple website](https://www.apple.com/macos/ventura/) shows availability 10.24.  I'll check to update throughout the day today.

mliberty1, 2022-10-30:
Joulescope UI version 0.10.8 now runs on macOS 13.  Note that the 0.10.x releases use a new C driver backend and add support for the new JS220.  It still supports the JS110.  You can download it [here](https://download.joulescope.com/joulescope_install/index_v2.html).  We do plan to make this (or a future version) available on the main download page by Thu, Nov 3.

We have tested UI 0.10.8 on both arm64 and x86_64, so it should work for you.  I am going to close this issue, but please reopen it if 0.10.8 does not run for you.

closed 2022-10-30 by mliberty1