jetperch/pyjoulescope_ui #168: [Bug]: Incorrect scaling in live view

NevynUK, 2022-10-31:
### Joulescope serial number

001736

### UI version

0.10.8

### What OS are you seeing the problem on?

macOS 13 amd64

### What happened?

I have connected a Meadow (microcontroller board) using the USB powered hub through to a MacBook Pro.  This board has nothing running except the OS so I am expecting low current draw.  Running the UI connected to the Joulescope shows the following:

[image]

Note the scale is 0 - 5A and the statistics also reflect this:

[image]

Note that the multimeter view is showing the correct average , minimum and maximum values.

### What was expected?

I am expected the board to average about 130mA with periodic peaks.  The following is a JLS file from a 5 minute run captured using the Python capture.py script:

[image]

This looks correct.

### How to reproduce?

1. ... Connect a low current microcontroller board to a computer through a JS110.
2. ... Start the Joulescope UI

mliberty1, 2022-10-31:
Thanks @NevynUK !  It sounds like the current range change filter is not being applied correctly.  I was able to duplicate the issue.

Here is a capture of the original JS110 EVK using 0.9.11:
[image]

And the same capture using 0.10.8:
[image]

Here are the current range settings:
[image]

We'll be investigating today!

mliberty1, 2022-10-31:
I tracked down the issue, which is actually in the joulescope driver [js110_sample_processor](https://github.com/jetperch/joulescope_driver/blob/main/src/js110_sample_processor.c).  The code was not correctly respecting the "n" and "m" samples_window settings.  We should have a new Joulescope UI release tomorrow.

NevynUK, 2022-11-01:
Thank you for the update.  Look forward to using the new release.

closed 2022-11-01 by mliberty1