jetperch/pyjoulescope_ui #169: [Bug]: Statistics for dual markers show NaN for mean and standard deviation

NevynUK, 2022-10-31:
### Joulescope serial number

001736

### UI version

0.10.8

### What OS are you seeing the problem on?

macOS 13 amd64

### What happened?

I have captured a data set and loaded this into the UI and added dual markers to the plot.  The mean and standard deviation are showing as NaN

[image]

### What was expected?

I would expect the mean and standard deviation for the samples between the two markers to be calculated

### How to reproduce?

1. ... Capture data using the command `python capture.py --duration 5 --jls 2 BoardStartup.jls`
2. ... Load the data set into the UI
3. ... Add dual markers

I can provide the capture file if necessary, be aware that the capture is 79 MByte.

NevynUK, 2022-10-31:
I have just re-run the test another 4 times and I'm seeing similar output.

[image]

[image]

[image]

I even tried releasing the reset button a lettle earlier in case it was an issue with data throughput:

[image]

The capture was run using the Python scripts and not the UI.

Also, in the captures above I killed everything that was not essential to the test so the only active applications were
* UI (not being used for capture only for showing the files)
* iTerm running the capture process using Python
* TextEditor - doing nothing

Only other thing that could complicate things is the fact I am running through a USB hub.  Only othre things connected are mouse and I made sure I did not move it during the capture.

NevynUK, 2022-10-31:
Moved the Joulescope to a laptop USB port so data is no longer coming from the hub but direct to the MacBook and similar profiles are being recorded.

NevynUK, 2022-10-31:
May have a theory about the cause of the data glitch - I think it may be a problem caused by our board configuring itself for USB comms.

I have just plugged the board into a USB power supply and the Joulescope into the laptop and there is no data drop out and the markers work as expected.

mliberty1, 2022-10-31:
That is interesting.  I understand that you are using the capture script for the recording.  So, the UI version should not impact the recording and sample drops.  The UI version will affect the NaNstatistics display for dual markers, which we will address.

For the sample drops, the recording version depends upon what you have installed for python.  What does this display on your computer?

    python3 -c "import joulescope; print(joulescope.__version__)"

Unless you did a `pip3 -U joulescope` very recently, I would expect it to show `0.9.11`.

Interesting theory about the target board causing the sample drops.  Bringing up a USB device can impact your Joulescope JS110 in two ways: power (brown-out) and data.  If bringing up the target board causes the USB VBUS to drop significantly, then your JS110 may brown out and reboot.  I think I am seeing about  0.15 seconds of drop in what you shared, which seems too fast for a JS110 reboot and USB reconnect.  

Alternatively, if your target board requests significant data communication or otherwise blocks up USB communications, then your JS110 may not have enough USB bandwidth to communicate samples resulting in sample drops.

You don't happen to have data from macOS 12 before updating to macOS 13 to see if this behavior has changed, would you?

NevynUK, 2022-11-01:
Running the Python command does show that I have version `0.9.11`

I was wondering if it was more of a software problem with the driver on the laptop.  So my first setup is as follows:

USB-C port -> USB Hub (not powered) -> USB Hub (mains powered) -> Joulescope + DUT

I then tried this setup:

USB-C port 1 -> USB Hub (not powered) -> USB Hub (mains powered) -> Joulescope
USB-C port 2 -> DUT

I saw the same behaviour in both cases.  I would have hoped that the mains powered hub would have prevented a brownout in both cases.  In the second case I would have thought the splitting of the setup would make it highly unlikely.

I was wondering if it may be the registration of our device as a USB modem could be causing the problem.  To be honest USB setup/operation on laptops is not my field of expertise.

This is the first time we have been running these sorts of test so I do not have previous data to compare the results to.  I do however have a lower spec Mac running a previous version of MacOS so I can set that up as a comparison.

NevynUK, 2022-11-01:
So just tried fired up the Mac Mini running Catalina (MacOS 10) and Joulescope UI 0.9.11.

I could not get the Python scripts installed for some reason.  Job for later.

However, I did get the UI up and running.  I ran the same scenario and the display froze part way through the capture.  I noticed that the USB indicator in the bottom right of the UI changed from green to red at the same time that the data capture paused.  When the data capture restarted the same image from above was displayed.

mliberty1, 2022-11-01:
The next time you start the Joulescope UI, you should be prompted to download Joulescope UI [0.10.9](https://www.joulescope.com/pages/downloads) (or newer).  This version fixes the original issue and will ignore NaNs when computing the dual marker statistics values.

This version uses the new v1 Joulescope driver as the backend.  This new driver may exhibit different capture behavior.  If you want to try it out, issue `pip3 install -U joulescope` from the command line and then `python3 -m joulescope capture out.jls`.  I am going to close this issue since 0.10.9 addresses the reported issue.  

We may not yet have a fully working solution for you that avoids dropped samples in your capture.  If you continue to have an issue with dropped samples, please comment below or open a new issue.

closed 2022-11-01 by mliberty1