jetperch/pyjoulescope_ui #171: Crash on macOS

mliberty1, 2022-11-07:
### Joulescope serial number

N/A

### UI version

0.10.9

### What OS are you seeing the problem on?

macOS 13 arm64 (M1 and newer), macOS 13 amd64, macOS 12 arm64 (M1 and newer), macOS 12 amd64

### What happened?

The Joulescope UI consistently crashes on macOS.

### What was expected?

The UI should not crash.

### How to reproduce?

1. Clear configuration: `rm "~/Library/Application Support/joulescope/config"`
2. Launch the Joulescope UI 
3. Select **View** → **Oscilloscope**
4. Click on any other active application
5. Click back on the Joulescope UI
6. The Joulescope UI immediately crashes

### Extra information

This crash occurs even *WITHOUT* a Joulescope JS110 or JS220 connected.  So, this is likely an issue with improper use of PySide6/Qt6.

mliberty1, 2022-11-07:
Further testing shows that you don't even need to click on another application.  Launching any dialog within the Joulescope UI has the same behavior.  For example, select **Help** → **Getting Started**.  The crash happens just after clicking **OK**.  Likewise, the same crash happens with the **Joulescope** → **Preferences...** dialog.

mliberty1, 2022-11-08:
We are actively working on tracking down this issue.  We do have an inconvenient workaround that seems reliable so far.  You can exit the Joulescope UI in the oscilloscope view, restart it, and then never switch to the multimeter view.  If you do need to switch, you have to close the Joulescope UI, reopen, select multimeter view, close the Joulescope UI, reopen, and then you are good to go.  Really, anytime you add or remove a widget, you need to exit and restart the Joulescope UI.

mrks115, 2022-12-02:
Any update here?

mliberty1, 2022-12-02:
Thanks for checking back @mrks115 .  I did an initial investigation, and I could not identify the actual cause of the crash.  I suspect that the code is not holding on to a PySide6 QWidget reference somewhere.  The existing widget lifetime management is definitely not as clean as it should be, which makes this extra suspicious.

We are rewriting how we manage widgets with the upcoming 1.0 release.  Since I was not able to find the actual root cause quickly and we have this inconvenient workaround, I have decided not to fix this in the 0.10.x line.  Unfortunately, the 1.0 release will still take time.  I apologize that the workaround is very inconvenient, but we'll get there!

mliberty1, 2023-03-10:
Fixed in 1.0.0 which runs well on macOS 13.2.1 arm64 and x86_64.  This release has **lots** of changes from 0.10.x, so it's definitely possible we have new issues with different causes.  I am going to close this issue.  If anyone experiences crashes on macOS or any other platform, please feel free to open a new issue!

closed 2023-03-10 by mliberty1