jetperch/pyjoulescope_ui #174: [Bug]: Joulescope UI crash with JS110

med-anis-bm, 2022-11-15:
### UI version

0.10.9

### What OS are you seeing the problem on?

Windows 10

### What happened?

i was using the joulescope UI v0.9.7 with the JS110.
I received the new JS220 so i updated the UI to the latest version **0.10.12** and it automatically crashes when using the JS110.
The UI does not crash when using the JS220 but it's less responsive than the previous version i used to use.

[image]

[image]

[image]

### What was expected?

 Keep the compatibility with the JS110

### How to reproduce?

1. ...connect a JS110 with the latest UI v0.10.12
2. ...
3. ...

mliberty1, 2022-11-15:
Hi @med-anis-bm - I suspect that you are duplicating issue #172.  You may have accidentally selected a huge buffer size which is causing your computer to swap RAM to disk.  

You can try manually adjusting this preference.

Alternatively, you can clear your preferences.  Close the Joulescope UI.  In Windows file explorer, navigate to %LOCALAPPDATA%\joulescope and delete all subdirectories.  Then open the Joulescope UI.  Does the problem persist?

mliberty1, 2022-11-15:
Compared to 0.9.x, we do know that the 0.10.x Joulescope UI has lower performance at the full sampling rate.  The new backend does not have the reduction implementation of the previous backend.  However, they have the same performance for downsampled data.  

We are working on an even faster reduction implementation that works for both full-rate and downsampled data.  It also will fix #172.

mliberty1, 2022-11-15:
Wow, 20 minutes is a big RAM buffer.  The JS220 requires half the RAM as the JS110.  Although both sample at 2 Msps, the JS220 downsamples to 1 Msps on the instrument while still keeping the 300 kHz bandwidth.

How much physical RAM does your PC have?

mliberty1, 2022-11-15:
For what it's worth, the v1 backend used by Joulescope UI 0.10.x requires 15 bytes per sample (plus a small amount of overhead).  So, at 2 Msps and 20 minutes, you would need:

15 B/sample * 2e6 samples/s * 60 seconds/minute * 20 minutes = 36 GB.

The JS220 at 1 Msps would *only* need 18 GB.

med-anis-bm, 2022-11-15:
i have 32GB of RAM.
That's unfortunate, we were using the joulescope for long acquisitions to see the behaviour of our sensors in extreme temperatures for example. The tests lasted hours and it was very well handled in the previous backend. 
We'll keep waiting for the fix to use the JS220. Until then, we're happy with the JS110 with the old backend so maybe we'll downgrade for now and see what happens. Thank you for the answers

mliberty1, 2022-11-15:
I am very surprised that the 0.9.x Joulescope UI handled a 20-minute buffer at full sample rate on a 32 GB machine!

Are you aware that you can use a 30-second RAM buffer and still record to disk for days?

med-anis-bm, 2022-11-15:
By launching the record, that's how i'm doing it. I'll gladly have any other suggestion 
[image]

med-anis-bm, 2022-11-16:
No, were're talking about the same thing. My issue is about the RAM buffer which is defined in File → Preferences → Device → setting → buffer_duration.
My previous comment which was about the record is just an answer to "Are you aware that you can use a 30-second RAM buffer and still record to disk for days?".  I showed you how i'm doing it **if** i wanted a 30s buffer but record for days.
The main issue is really about the RAM buffer that crushes when 20 minutes duration is selected.

mliberty1, 2022-11-16:
Ok, so we are both talking about the RAM buffer.  I think then that this issue is a duplicate of #172.

With the 0.9.x Joulescope UI, each JS110 sample required 13 bytes:
13 B/sample * 2e6 samples/s * 60 seconds/minute * 20 minutes = 31.2 GB.

So, your computer with 32 GB RAM may have just been able to get by without too much swapping to disk.  However, it was definitely marginal.

The 0.10.x Joulescope UI does not include the reductions optimization, which requires the computer to perform more computations.  This additional workload combined with a slightly larger buffer appears to push your computer beyond its capabilities so that it cannot keep up with the incoming data.  We will be improving buffering performance in the not too distant future.  However, we will also restrict the buffer to physical RAM, which may prevent you from even selecting a 20 minute buffer at the full 2 Msps for the JS110.

In the meantime, here are some options:
1. Reduce the RAM buffer size to 10 minutes, which is still great!
2. Continue to use the Joulescope 0.9.x software for when you want the 20 minute buffer.
3. Upgrade the RAM on your PC.

I am going to close this issue since it is a duplicate.  However, please feel free to comment more or reopen it if I have missed anything.

mliberty1, 2022-11-16:
Duplicate #172

closed 2022-11-16 by mliberty1