jetperch/pyjoulescope_ui #175: [Bug]:

med-anis-bm, 2022-11-15:
### UI version

0.10.9

### What OS are you seeing the problem on?

Windows 10

### What happened?

I'm using the JS220 with the latest UI version **0.10.12**.
In oscilloscope mode, the voltage value is always 2.2V knowing that i i'm injecting 3V. I verified with an external voltmeter and the voltage in input and output of the joulescope is 3V.
[image]

When switching to multimeter view, the UI displays the correct voltage.
[image]

### What was expected?

Oscilloscope should display the correct voltage

### How to reproduce?

1. ...Select oscilloscope view
2. ...Check the voltage value
3. ...

mliberty1, 2022-11-15:
Hi @med-anis-bm - Did you select the fixed 2V range?  Like this:

[image]

Note that the Multimeter View and Oscilloscope View have their own defaults and will switch this setting.

med-anis-bm, 2022-11-15:
Hi, No im selecting 15V range 
[image]

mliberty1, 2022-11-15:
It looks like you have stopped streaming.  Could you start streaming?  Is it then correct?

If you still see the issue, could you try toggling to 2V then back to 15V?  Any difference in the reading?

med-anis-bm, 2022-11-15:
Switching to 2V and then going back to 15V resolved the issue, Thank you

mliberty1, 2022-11-15:
Great!  I tried playing around, and I think that I figured out how you arrived at this issue, which is actually a bug.  The issue is switching from the JS110 to the JS220.  Here is how to duplicate:

1, Connect both a JS110 and a JS220 to the same host computer and start the UI,
2. Select a JS110
3. Set **Voltage Range** => **5V**
4. Select a JS220
5. The **Voltage Range** now shows 15V (since 5V is not valid), but the UI still sets the JS220 for the more sensitive 2V range.
6. Select **2V** then **15V** fixes.

med-anis-bm, 2022-11-15:
> Great! I tried playing around, and I think that I figured out how you arrived at this issue, which is actually a bug. The issue is switching from the JS110 to the JS220. Here is how to duplicate: [...]

I confirm, That was exactly the case

mliberty1, 2023-03-10:
Fixed by 1.0.0 which has separate settings for JS110's and JS220's.

closed 2023-03-10 by mliberty1