jetperch/pyjoulescope_ui #176: [Bug]: Charge / Energy numbers jumping up and down

itsonlyaudio, 2022-12-19:
### Joulescope serial number

000518

### UI version

0.10.9

### What OS are you seeing the problem on?

Linux Ubuntu amd64 (PC)

### What happened?

In my gui (0.10.12) the number of mAh and mWh jumps around as soon as it goes over ~2144. It also appears in the logged data. Graph for illustration:

[image]

Although the GUI couns in (m)Ah and the log in As (I log once per second) the error is in both, pointing towards the integration algorithm.
When I integrate the current over time myself in a spreadsheet the numbers are correct:

[image]

### What was expected?

Correct energy / power integration

### How to reproduce?

1. let energy accumulate over 2144 As / 2144 Ws
2. see if numbers jump to ~6143 As / Ws suddenly thereafter
3. scratch head

### Extra information

Logging spreadsheet attached: file:///home/remco/Documents/joulescope/20221219_140403%20battery%20moto.xlsx

mliberty1, 2022-12-19:
Well, that's weird.  With the new joulescope_driver, we use 128-bit integer math in C to accumulate these values to maintain sufficient precision.  The 0.9.x series UI used the python-based pyjoulescope driver with python integers.  

Possible places for cause of this issue:
1. [js220_i128](https://github.com/jetperch/joulescope_driver/blob/main/src/js220_i128.c)
2. [js110_stats](https://github.com/jetperch/joulescope_driver/blob/main/src/js110_stats.c#L122)
3. [python binding](https://github.com/jetperch/joulescope_driver/blob/main/pyjoulescope_driver/binding.pyx)
4. Joulescope UI (but unlikely)

I'll see what I can find...

mliberty1, 2022-12-19:
The Excel file shows the jump from 2146.97 to 6442.95, which is a difference of 4296.  These values are stored as fixed-point with representation 87Q31, but then divided by the number of samples per window.  The JS110 is 2 Msps.  This is interesting:

    ((1 << 64) >> 31) / 2e6  =>  4295

So, the issue is likely related to "overflowing" into the upper 64-bits of the 128-bit word.  Alternately, it could be the 63rd bit of the lower word being used as a sign bit over the 2 Hz window.

Note that the JS110 is still much less tested under the Joulescope UI 0.10.x, which is why the Joulescope UI 0.9.x does not yet prompt customers to update yet.  If you are using a JS110, then the quickest fix is to downgrade to Joulescope UI 0.9.x while we continue to investigate.

itsonlyaudio, 2022-12-19:
OT: I had to upgrade to 0.10 because 0.9.x stopped working when I switched ubuntu over to wayland (moving from nvidia to radeon).

mliberty1, 2022-12-20:
This issue should be resolved by 0.10.13, now available to download from our [download page](https://download.joulescope.com/joulescope_install/index_v2.html) as a beta.

I confirmed that the joulescope_driver was fixed overnight.  I just started a test to confirm the Joulescope UI with 721 mA @ 0.98V, and it will take close to an hour to pass ~2200 joules to confirm the problem is fixed.

mliberty1, 2022-12-20:
Now working great on Ubuntu 22.04 LTS.  I am going to close the issue, but please reopen if you still see an issue.

itsonlyaudio, 2022-12-20:
I can confirm that 0.10.13 fixes the problem.

[image]

Thanks so much!

closed 2022-12-20 by itsonlyaudio