jetperch/pyjoulescope_ui #177: Oscilloscope view of JLSv2 | Statistics for dual marker add delta energy

ClimbTheWorld, 2023-01-23:
### UI version

0.10.9

### Your idea

Hi,
When openening a JLSv2 file in the oscilloscope view and using a marker the delta energy is not shown, where it is in JLSv1.

mliberty1, 2023-01-23:
Thanks for reporting this issue, @ClimbTheWorld .  We are currently working on a major Joulescope UI update, and we will ensure that this works in the updated UI.  The current plan is to have an alpha version in early February (~2 weeks from today).

The workaround for 0.10.x is to manually multiply the average power by the total time.

mliberty1, 2023-05-16:
Confirm fixed in 1.0.14 alpha (beta pending ongoing testing).  Download here:

https://download.joulescope.com/joulescope_install/index_v2.html

closed 2023-05-16 by mliberty1