jetperch/pyjoulescope_ui #178: [Question]: feature/1_0 does not start on macOS, rccs not compiled.

schneider-engineering, 2023-02-03:
### UI version

0.10.9

### Your question

Somehow the CustomBuildQt function in setup.py seems never to be called. Installed with `pip install -e .` Am I holding this wrong? Feel free to close if you do not want any questions IRT feature/1_0 as of now :)

MacOS 12.6.1

```
--------------------
Exception on Joulescope UI startup. 

Traceback (most recent call last):
  File "/Users/user/path/pyjoulescope_ui/joulescope_ui/main.py", line 298, in run
    resources = load_resources()
  File "/Users/user/path/pyjoulescope_ui/joulescope_ui/resources.py", line 33, in load_resources
    b = pkgutil.get_data(*r)
  File "/opt/homebrew/Cellar/python@3.10/3.10.8/Frameworks/Python.framework/Versions/3.10/lib/python3.10/pkgutil.py", line 639, in get_data
    return loader.get_data(resource_name)
  File "<frozen importlib._bootstrap_external>", line 1073, in get_data
FileNotFoundError: [Errno 2] No such file or directory: '/Users/user/path/pyjoulescope_ui/joulescope_ui/resources.rcc'

info = {
  "joulescope": {
    "ui_version": "1.0.0"
  },
  "platform": {
    "name": "darwin",
    "python_version": "3.10.8 (main, Oct 13 2022, 09:48:40) [Clang 14.0.0 (clang-1400.0.29.102)]",
    "platform": "macOS-12.6.1-arm64-arm-64bit",
    "processor": "arm",
    "executable": "/Users/user/path/pyjoulescope_ui/.venv/bin/python",
    "frozen": false
  }
}
--------------------
```

```
Package             Version   Editable project location
------------------- --------- -----------------------------------------------
appdirs             1.4.4
appnope             0.1.3
certifi             2022.12.7
charset-normalizer  3.0.1
fs                  2.4.16
idna                3.4
joulescope          1.0.16
joulescope-ui       1.0.0     /Users/user/path/pyjoulescope_ui
Markdown            3.4.1
numpy               1.24.1
packaging           23.0
pip                 22.2.2
psutil              5.9.4
pyjls               0.4.3
pyjoulescope-driver 1.1.4
pymonocypher        3.1.3.1
PyOpenGL            3.1.6
pyperclip           1.8.2
pyqtgraph           0.13.1
PySide6             6.4.0.1
PySide6-Addons      6.4.0.1
PySide6-Essentials  6.4.0.1
PySide6-QtAds       3.8.3.1
python-dateutil     2.8.2
QtPy                2.3.0
requests            2.28.2
setuptools          65.4.1
shiboken6           6.4.0.1
six                 1.16.0
urllib3             1.26.14
```

### What OS are you using?

macOS 12 arm64 (M1 and newer)

mliberty1, 2023-02-03:
I haven't figured out how to make this automatic.  The release packages you install from pypi include the qt resources (no qt resource build required).  When installing from source or developing, you need to manually build them first:

    python3 setup.py qt

Ideally, pip & setuptools would know the difference and do the right thing in both cases.  I purchased [Publishing Python Packages](https://www.manning.com/books/publishing-python-packages) last week.  Maybe it will have some tips when I get around to reading it!

Does this workaround work for you?

schneider-engineering, 2023-02-04:
Thank you for your hint. This works fine for me. I was (mistakenly) under the impression that somehow this must have been called automatically. See #179 for crude way to call this automatically.

closed 2023-02-04 by schneider-engineering