jetperch/pyjoulescope_ui #182: UI Hangs

jkhax0r, 2023-04-13:
### Joulescope serial number

001052

### UI version

0.10.13

### What OS are you seeing the problem on?

Windows 11

### What happened?

I tried out the new Alpha 1.x.x UI.  Everything seems to work.  Then went back to old and it just hangs.  Seems to be for about 2 minutes locking computer up almost completely and then quits.  But still doesn't work and when I toggle disabled and enabled again it locks up again for a few minutes.  Haven't gotten joulescope to work again on the old UI.  I think new UI will work fine.

### What was expected?

Older UI works (0.10.13)

### How to reproduce?

Above

mliberty1, 2023-04-13:
Hi @jkhax0r - Sorry to hear that this is not working for you.  I personally switch between the different versions without an issue, so it is definitely possible.

Here are some possible causes for what you see:

1. Issue #174 which duplicates #172?  You can try clearing your 0.10.x configuration, which is %LOCALAPPDATA%\joulescope\config\joulescope_config.json.

2. Another program (including the Joulescope 1.0 alpha) still has the Joulescope open.  The UI 0.10.x does not handle this case well.

3. Something about UI 0.10.x is unhappy about the JS110 state.  While this is unlikely, you can try unplugging and replugging your JS110 to see if this helps.

Do any of these seem to apply to your situation?  If you want, you can post the logs here or email them to support **at** joulescope **dot** com.  If you email them, please reference this issue, #182.

jkhax0r, 2023-04-13:
Matt, Thanks for the quick reply as usual.  I tried (3) but (2) could be the case too.  Also I remember doing (1) once before probably a year or 2 back, so that may be it.  

Mostly unrelated, but it's what led me to upgrading/downgrading...  is there an easy way to see f/w version or force an update?  Can it be downgraded?  Or do none of the f/w versions matter much.  I do custom USB commands (I think based on your C example) so thought this may matter since we were seeing some out-of-bounds numbers.  

BTW, my root issue is that my manufacturer is failing tests where during idle periods of low current (2uA), it basically reads negative current with lots of ripple.  I suspect an inferior power supply/noise even though they have the same brand as I do.  

Here's a snapshot of my setup at home (left) vs factory (right) if you have any quick tips to help.  Would you agree its power supply ripple and low currents?  My only doubt is that when I change my JS110 from 2Mhz to e.g. 10Khz, then I can see a lot of that ripple too.  Even the ripple after the wakeup event (t=350ms on left) changes drastically depending on the JS110 sample speed.  

[image]

mliberty1, 2023-04-19:
We just released Joulescope UI 1.0.9.  Could you update and see if this problem is fixed?

If you still have 1.0.8 installed, you can launch it and you will be prompted to update.  Otherwise, you can [download](https://download.joulescope.com/joulescope_install/index_v2.html) and install.

jkhax0r, 2023-04-23:
I've added an LM317 (cheap LDO) regulator from Amazon, added after my cheap switching lab supply, and that seems to have helped A LOT.  Note that I had purposely bought about 3 cheap lap power supplies and this was the worst that I have been using to try to fix this.  

I still have the same sort of issues during large transitions (e.g. before I wakeup and then before charging an inductor for a pulse).   But it seems this is largely related to my slow sample rate - I had setup 10Khz just to lower processing time.  I'm guessing the transitions between modes create some sort of instability that is better at 2Mhz.  It's likely the inductance in the wire/circuits that is causing some oscillations that sort of resonate at the lower frequencies and exacerbated by auto current switching.

A series R did help a bit... and I have tried many mF on my LM317 too to just make that is as ideal as possible - but mostly just having the LM317 and its stock 47uF was the biggest step.

I had setup a current envelope for my tester at 10khz, so I'll try this 2Mhz path and then simply average in my software down to 10khz to see if I can get around.  The glitches at 2Mhz are on the order of 2 samples, so I suspect down sampling this may fix everything.  

I also started designing a little LDO on a "front-panel" that I'll share with you if I ever finish.  Its in Altium though.  Just a variable pot to dial between 1 and 5V and to help make sure my manufacturing testers are as controlled as possible.

mliberty1, 2023-04-24:
Great to hear that the setup is working better!

Your JS110 always samples at 2 Msps, and the current range switching is always performed on the 2 Msps data.  The JS110's downsampling to 10 kHz is done on the host in software.  I would not expect to see any difference in electrical behavior due to current range switching based upon the selected output sampling rate.

The JS110 autoranges in 1 µs, but remains "blind" for about 5 µs.  If those 2 sample long glitches cause a JS110 current range change and the glitches contain most of the energy, then the JS110 measurements may have significantly reduced accuracy. 

The JS220's new Enwavify™ technology adds additional hardware so that it can measure through the current range changes with no blind spots.  The accuracy until the new current range path settles is the same as the 10 A range, which is ±0.25% ±1.50 mA.   The JS220 has significantly improved RF and high-frequency rejection compared to the JS110, so it behaves much better with fast transients like this.

Look forward to seeing what you put together for the front panel.  If you want to make it computer controlled, I have been very happy using the Raspberry Pi Pico with micropython for the JS220 manufacturing equipment.  It's easy to control from python on the host using pyboard without any extra device firmware.  You can write micropython code for anything that needs to be faster.  For an example and related info, see the [JS220 EVK1](https://github.com/jetperch/js220_evk1).

You could use a Pico + DigiPot for this, but I am not sure how the mechanicals of mounting a Pico on a JS220 front panel would work out....

mliberty1, 2023-05-31:
Hi @jkhax0r and sorry for the delay in getting back to you!  This got buried in my email as I did not immediately connect the question to the topic.  I finally went back and reread the entire chain.

The oscillations in the current plot are expected at 10 kHz, and they look much worse viewed with a logarithmic scale.  Here is what you posted on a linear scale:
[image]

When Joulescopes downsample, they apply an finite impulse response (FIR) filter first to avoid aliasing.  The exact filtering/downsampling filters & stages depends on the input frequency and output frequency.  You can find the exact code with coefficients [here](https://github.com/jetperch/joulescope_driver/blob/main/src/downsample.c).

Now, sharp edges in the time domain require high measurement bandwidth to represent them.  As you reduce the measurement bandwidth, you will observe more and more "ringing" around sharp edges in the time domain.  That is exactly what you are seeing here.  Interestingly, you can see the same "ringing" in the current range, even though the current range is switching the same as before.

If you want to research more, you can check out the Fourier series of a square wave.  A Google search for "square wave fourier series" turns up a bunch of decent guides.

Downsampling trades storage space for detail in the time domain.  The effective measurement bandwidth ends up being about 0.4 of the selected downsample frequency.  As you downsample below the original system bandwidth, the downsampling process will distort the original signal, flattening out peaks and introducing ringing around edges.  This effect is common to any time you reduce measurement bandwidth on any instrument!

Does this make sense and explain what you are seeing?

I am going to close this issue.  If you would like to discuss this further, this would make a great "Technical Discussion" topic over on the [Joulescope forum](https://forum.joulescope.com/).

closed 2023-05-31 by mliberty1